gekkekoe/esphome-ecodan-hp #8: Slow serial makes queue increase, loose commands

jessevl, 2024-05-11:
I've tried using this integration for ESPHome with my FTC6-based Mitsubishi Ecodan heatpump. However, I'm unable to process any commands as the serial interface seems too slow and thus will never get to fulfill the commands in the (ever increasing) queue The hardware used is multiple different M5StickC and M5stickC-Plus units, they all behave the same.

Changing the polling frequency did not change the issue. I'm fairly sure the TX/RX pins are working correctly, is the below output even possible if it isn't?

Relevant esphome config:
```
substitutions:
  name: ecodan-heatpump
  friendlyName: Ecodan Heatpump

esphome:
  name: ${name}
  friendly_name: ${friendlyName}

external_components:
  - source: github://gekkekoe/esphome-ecodan-hp@main
    components: [ ecodan ]
    refresh: always

packages:
  remote_package:
    url: https://github.com/gekkekoe/esphome-ecodan-hp/
    ref: main
    refresh: always
    files: [ 
            confs/esp32.yaml,
            confs/zone1.yaml,
            ## enable if you want to use zone 2
            #confs/zone2.yaml,
            ## enable label language file
            confs/ecodan-labels-en.yaml,
            #confs/ecodan-labels-nl.yaml,
            #confs/server-control.yaml,
            confs/debug.yaml,
           ]

api:
  reboot_timeout: 0s

web_server:
  port: 80

# enable logger
logger:
  baud_rate: 0
  logs:
    esp32.preferences: DEBUG
    sensor: DEBUG
    text_sensor: DEBUG
    switch: DEBUG
    button: DEBUG
    number: DEBUG
    climate: DEBUG
    component: ERROR

ota:
  password: "**"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Ecodan Fallback Hotspot"
    password: "YCnaBILWnkhA"

captive_portal:

ecodan:
[... 198 lines cut ...]
> [17:40:24][D][sensor:094]: 'sh_flow_temp_target': Sending state 20.00000 °C with 1 decimals of accuracy
> [17:40:25][D][sensor:094]: 'flow rate': Sending state 19.00000 l/m with 0 decimals of accuracy
> [17:40:25][D][text_sensor:064]: 'booster heater status': Sending state 'Off'
> [17:40:25][D][text_sensor:064]: 'immersion heater status': Sending state 'Off'
> [17:40:26][D][text_sensor:064]: 'water pump status': Sending state 'On'
> [17:40:26][D][text_sensor:064]: '3-way valve status': Sending state 'Off'
> [17:40:26][I][ecodan.component:056]: Could not publish state of sensor 'status_water_pump_2' with value: 'On'
> [17:40:26][I][ecodan.component:056]: Could not publish state of sensor 'status_three_way_valve_2' with value: 'On'
> [17:40:27][D][sensor:094]: 'Heap Free': Sending state 186540.00000 B with 0 decimals of accuracy
> [17:40:27][D][sensor:094]: 'Loop Time': Sending state 110.00000 ms with 0 decimals of accuracy
> [17:40:27][D][sensor:094]: 'operating runtime': Sending state 1544.00000 h with 0 decimals of accuracy
> [17:40:29][D][text_sensor:064]: 'In1 thermostat status': Sending state 'On'
> [17:40:29][D][text_sensor:064]: 'In6 thermostat 2 status': Sending state 'Off'
> [17:40:30][D][sensor:094]: 'boiler flow temp': Sending state 25.00000 °C with 1 decimals of accuracy
> [17:40:30][D][sensor:094]: 'boiler return temp': Sending state 25.00000 °C with 1 decimals of accuracy
> [17:40:31][D][sensor:094]: 'feed temp': Sending state 14.50000 °C with 1 decimals of accuracy
> [17:40:31][D][sensor:094]: 'return temp': Sending state 19.00000 °C with 1 decimals of accuracy
> [17:40:31][D][sensor:094]: 'DHW current temp': Sending state 50.00000 °C with 1 decimals of accuracy
> [17:40:32][D][sensor:094]: 'Heap Free': Sending state 186540.00000 B with 0 decimals of accuracy
> [17:40:32][D][sensor:094]: 'Loop Time': Sending state 112.00000 ms with 0 decimals of accuracy
> [17:40:32][D][sensor:094]: 'Zone 1 room temp': Sending state 24.50000 °C with 1 decimals of accuracy
> [17:40:32][I][ecodan.component:045]: Could not publish state of sensor 'z2_room_temp' with value: '0.000000'
> [17:40:32][D][sensor:094]: 'outside temp': Sending state 33.00000 °C with 0 decimals of accuracy
> [17:40:32][D][sensor:094]: 'refrigerant temp': Sending state 14.50000 °C with 1 decimals of accuracy
> **[17:40:33][I][ecodan.component:290]: command queue was not empty when queueing status query: 20**`

gekkekoe, 2024-05-12:
Thanks for the detailed report, that helps.
I'm missing the compressor frequency response (also defrost stuff, but this was the first that caught my attention).
It seems that that cmd is not sent to the heat pump. It should be the 3rd cmd. But also defrost state is missing. Could be that it was before it started logging.

I have changed the get status to clear pending get before added new get status cmd. These will be added at the end of the queue ensuring that set cmd actually get's executed.

your loop time is similar to mines, so the esp32 should have no issues with processing the cmd. I don't think this is the solution for the root cause, bit it will help. let me know if it improves things

gekkekoe, 2024-05-12:
it seems that the 0x26 is coming back 'out of order' and way later than expected. This is quite odd since my ftc6 returns it request almost instantly. What heatpump model do you have? could also be ftc6 firmware.

jessevl, 2024-05-12:
The indoor unit is an erst30d-vm2ed. I can't find the firmware version.

gekkekoe, 2024-05-12:
try this branch and report back, it will allow more time for 'slower' units without affecting 'faster' units
https://github.com/gekkekoe/esphome-ecodan-hp/tree/test_allow_more_wait
I cannot trigger the conditions locally, so you guys have to test it :)

jessevl, 2024-05-13:
Seems to work okay, but still get queuing and discarded cmd's. FTC6 seems very slow to respond at times.

> [12:05:51][D][debug:079]: ESPHome version 2024.4.2
> [12:05:51][D][debug:083]: Free Heap Size: 168392 bytes
> [12:05:51][D][debug:113]: Flash Chip: Size=4096kB Speed=40MHz Mode=DIO
> [12:05:51][D][debug:162]: Chip: Model=ESP32, Features=EMB_FLASH,WIFI_BGN,BLE,BT, Cores=2, Revision=1
> [12:05:51][D][debug:170]: ESP-IDF Version: v4.4.2
> [12:05:51][D][debug:175]: EFuse MAC: **
> [12:05:51][D][debug:275]: Reset Reason: Software Reset CPU
> [12:05:51][D][debug:320]: Wakeup Reason: Unknown
> [12:05:52][D][sensor:094]: 'heating delivered': Sending state 0.01000 kWh with 2 decimals of accuracy
> [12:05:52][D][sensor:094]: 'cool delivered': Sending state 58.88000 kWh with 2 decimals of accuracy
> [12:05:52][D][sensor:094]: 'dhw delivered': Sending state 11.25000 kWh with 2 decimals of accuracy
> [12:05:52][D][sensor:094]: 'heating cop': Sending state 0.00000  with 2 decimals of accuracy
> [12:05:52][D][sensor:094]: 'cool cop': Sending state 4.37770  with 2 decimals of accuracy
> [12:05:52][D][sensor:094]: 'dhw cop': Sending state 4.24528  with 2 decimals of accuracy
> [12:05:53][D][sensor:094]: 'controller_version': Sending state 3.00000  with 0 decimals of accuracy
> [12:05:54][D][sensor:094]: 'Heap Free': Sending state 186928.00000 B with 0 decimals of accuracy
> [12:05:54][D][sensor:094]: 'Loop Time': Sending state 210.00000 ms with 0 decimals of accuracy
> [12:05:57][D][text_sensor:064]: 'heat source': Sending state 'heatpump'
> [12:05:59][D][sensor:094]: 'Heap Free': Sending state 186688.00000 B with 0 decimals of accuracy
> [12:05:59][D][sensor:094]: 'Loop Time': Sending state 17.00000 ms with 0 decimals of accuracy
> [12:06:01][D][text_sensor:064]: 'controller version': Sending state 'FTC6'
> [12:06:04][D][sensor:094]: 'Heap Free': Sending state 186928.00000 B with 0 decimals of accuracy
> [12:06:04][D][sensor:094]: 'Loop Time': Sending state 17.00000 ms with 0 decimals of accuracy
> [12:06:05][D][sensor:094]: 'Esp Uptime': Sending state 521.17700 s with 0 decimals of accuracy
> [12:06:06][D][text_sensor:064]: 'defrost state': Sending state 'Off'
> [12:06:08][D][sensor:094]: 'compressor frequency': Sending state 34.00000 Hz with 2 decimals of accuracy
> [12:06:08][D][sensor:094]: 'WiFi Signal dB': Sending state -79.00000 dBm with 0 decimals of accuracy
> [12:06:08][D][sensor:094]: 'WiFi Signal %': Sending state 42.00000  % with 0 decimals of accuracy
> [12:06:09][D][sensor:094]: 'Heap Free': Sending state 186412.00000 B with 0 decimals of accuracy
> [12:06:09][D][sensor:094]: 'Loop Time': Sending state 118.00000 ms with 0 decimals of accuracy
> [12:06:09][D][sensor:094]: 'heat_source': Sending state 0.00000  with 0 decimals of accuracy
> [12:06:10][D][sensor:094]: 'output power': Sending state 3.00000 kW with 3 decimals of accuracy
> [12:06:11][D][sensor:094]: 'estimated output power': Sending state 3.13500 kW with 3 decimals of accuracy
> [12:06:12][D][sensor:094]: 'Zone 1 room setpoint value': Sending state 17.50000 °C with 1 decimals of accuracy
> [12:06:12][I][ecodan.component:045]: Could not publish state of sensor 'z2_room_temp_target' with value: '20.000000'
> [12:06:12][D][sensor:094]: 'Zone 1 H/C setpoint value': Sending state 15.00000 °C with 1 decimals of accuracy
> [12:06:12][I][ecodan.component:045]: Could not publish state of sensor 'z2_flow_temp_target' with value: '20.000000'
> [12:06:12][D][sensor:094]: 'legionella prevention temp': Sending state 65.00000 °C with 1 decimals of accuracy
> [12:06:12][D][sensor:094]: 'DHW max temp drop': Sending state 10.00000 °C with 1 decimals of accuracy
> [12:06:13][D][sensor:094]: 'Zone 1 room temp': Sending state 25.50000 °C with 1 decimals of accuracy
> [12:06:13][I][ecodan.component:045]: Could not publish state of sensor 'z2_room_temp' with value: '0.000000'
> [12:06:13][D][sensor:094]: 'outside temp': Sending state 26.00000 °C with 0 decimals of accuracy
> [12:06:13][D][sensor:094]: 'refrigerant temp': Sending state 15.00000 °C with 1 decimals of accuracy
> [12:06:14][D][sensor:094]: 'Heap Free': Sending state 186408.00000 B with 0 decimals of accuracy
> [12:06:14][D][sensor:094]: 'Loop Time': Sending state 113.00000 ms with 0 decimals of accuracy
> [12:06:14][D][sensor:094]: 'feed temp': Sending state 15.50000 °C with 1 decimals of accuracy
> [12:06:14][D][sensor:094]: 'return temp': Sending state 18.50000 °C with 1 decimals of accuracy
> [12:06:14][D][sensor:094]: 'DHW current temp': Sending state 64.50000 °C with 1 decimals of accuracy
> [12:06:16][D][sensor:094]: 'boiler flow temp': Sending state 25.00000 °C with 1 decimals of accuracy
> [12:06:16][D][sensor:094]: 'boiler return temp': Sending state 25.00000 °C with 1 decimals of accuracy
> [12:06:17][D][text_sensor:064]: 'In1 thermostat status': Sending state 'On'
> [12:06:17][D][text_sensor:064]: 'In6 thermostat 2 status': Sending state 'Off'
> [12:06:18][D][sensor:094]: 'operating runtime': Sending state 1568.00000 h with 0 decimals of accuracy
> [12:06:19][D][sensor:094]: 'Heap Free': Sending state 186412.00000 B with 0 decimals of accuracy
> [12:06:19][D][sensor:094]: 'Loop Time': Sending state 109.00000 ms with 0 decimals of accuracy
> [12:06:19][D][text_sensor:064]: 'water pump status': Sending state 'On'
> [12:06:19][D][text_sensor:064]: '3-way valve status': Sending state 'Off'
[... cut ...]
> [12:06:50][D][sensor:094]: 'flow rate': Sending state 18.00000 l/m with 0 decimals of accuracy
> [12:06:50][D][text_sensor:064]: 'booster heater status': Sending state 'Off'
> [12:06:50][D][text_sensor:064]: 'immersion heater status': Sending state 'Off'
> [12:06:50][D][text_sensor:064]: 'defrost state': Sending state 'Off'
> [12:06:51][D][sensor:094]: 'compressor frequency': Sending state 34.00000 Hz with 2 decimals of accuracy
> [12:06:52][D][sensor:094]: 'heat_source': Sending state 0.00000  with 0 decimals of accuracy

gekkekoe, 2024-05-13:
did you check that the discarding happens once +- a minute ? else you might still have linked the old code (you need to adjust the yaml to use the correct branch, or retrieve it locally and build locally)

jessevl, 2024-05-13:
That sounds about right, I got the "command queue not empty" every 30 seconds and the "discarding pending cmd" every 75 seconds. With the now latest version on the feature branch I get no errors anymore, and data seems to generally come in much more steadily. Awesome work. Much appreciated!

gekkekoe, 2024-05-13:
merged into main, closing

closed 2024-05-13 by gekkekoe