gekkekoe/esphome-ecodan-hp #9: Change DHW setpoint not working

DD-bep, 2024-05-11:
Hi, great job with this project!
I encountered a problem: changing the setpoints of both DHW and Zone 1 H/C has no effect, the DHW setpoint value and the Zone 1 H/C setpoint value ​​do not change.
Thank you

gekkekoe, 2024-05-12:
A bit more info would be helpful. Setting works for me and other users, so wondering why it does not work for you. can you provide more info (what heatpump model, in what mode when setting, do you see the logging msg that it updates new values, did you wait 30s, are other update/set  cmd working)?

also keep in mind that update sent to the heatpump can take up to 30s to become visible. 

There's another report with m5stickC that is having issues with sending update, and i've made changes to improve that. You could try the latest and see if it improves. It might be related

DD-bep, 2024-05-12:
Sure! The heat pump is an ecodan puhz-sw200yka (with a hydrobox unit that has an FTC5 controller) and I'm using an esp32 devkit. 
I updated the firmware and now I can change the DWH and Zone1 setpoints, great!
However now all energy related values (DHW delivered, DHW consumed, DHW cop, estimated output power, ecc) remain at unknown.
I don't know if that helps, but I checked the logs and about every 15 seconds it repeats:
```
[W][ecodan.component:296]: discarding pending cmd: 66
[W][ecodan.component:296]: discarding pending cmd: 91
```

Thank you

gekkekoe, 2024-05-12:
try this branch and report back, it will allow more time for 'slower' units without affecting 'faster' units
https://github.com/gekkekoe/esphome-ecodan-hp/tree/test_allow_more_wait
I cannot trigger the conditions locally, so you guys have to test it :)

roelzadh, 2024-05-13:
Seems om my FTC5 that H/C is changing the room temperature setpoint of zone 1 instead of the H/C target.

DHW setpoint seems to work as it should here.

gekkekoe, 2024-05-13:
merged into main, closing

closed 2024-05-13 by gekkekoe