gekkekoe/esphome-ecodan-hp #10: Wemos d1 mini, energy meter advice

Thebasestar, 2024-05-16:
Hi, first of all thanks to gekkekoe for this development, was waiting for something like this from 2 years, i have a few wemos d1 mini used for another project (  https://github.com/geoffdavis/esphome-mitsubishiheatpump )to control the mitsubishi aircon split : , can i use it for this ? 

Another request is that my installation is missing the energy monitor inside the ecodan, can someone guide me showing where they should be installed , whitch line to meter? My setup is made by a PUMY p112vkm5+ERST20C VM2DR1(FTC6)+3 split all r410.
I was plannig to add 2 SDM120 Modbus MID 45A or the model with clamp, are they ecquivalent for this use?

Tnks,
Base

gekkekoe, 2024-05-16:
The 8266 chips don't support free rtos via esphome, so that will not work (I use it for the background task). Get the atom s3 lite.

Why don't you get one of those wifi energy meters like https://www.homewizard.com/shop/wi-fi-kwh-meter-3-phase/ for 3 phase. And just put it before the powerline to your outdoor unit?

Thebasestar, 2024-05-16:
Ok i'll buy an atom, the meter you suggest is much more expensive than the sdm120( around 30€ each on aliex )...for what i've read the ecodan need 2 meter, one for the outdoor unit and one for the indoor (ftc+immersive heater)
[MMSP_APPLICATION_GUIDE_18_1.pdf](https://github.com/gekkekoe/esphome-ecodan-hp/files/15338738/MMSP_APPLICATION_GUIDE_18_1.pdf) with theese 2 meters installed the ftc should be able to determinate COP, sdm120 got pulse for the ftc6 and also modbus that i can eventually use straight on homeassistant. Do you have just one in your?

gekkekoe, 2024-05-16:
I only have one, since I only use the booster heather for the 2nd stage of a legionella run only I don't need a meter for the booster element. (it's disabled for heating and regular dhw).

Thebasestar, 2024-05-17:
Well i do the same, heater only for legionella, dont know if the second meter worth the price as it will only give the energy used by the ftc6 and derivates like pumps... still have to understand if my outdoor unit is powered from the inside one or the opposite

gekkekoe, 2024-05-17:
the outdoor powers the inside. this can be verified by cutting the power to the outdoor unit. the indoor will also power off.
only the booster element is powered separately

closed 2024-05-17 by gekkekoe