gekkekoe/esphome-ecodan-hp #14: No issue, more of a question.

bjornenvall, 2024-05-24:
Do you know if it’s possible to have the WiFi MELCloud thing connected at the same time?

gekkekoe, 2024-05-24:
I have beta software that supports that, but it will not work on the plain atom s3 (I used that setup to reverse engineer cmds).
You need to solder to get it working. I have it locally by splitting the 12v (procon won't power on with 5v), and wire another set of rx/tx pins to the esp and proxy the data between heatpump and wifi adapter/procon, but that would make this project hard to use, and that was not the goal.

I also see no real use for melcloud when you have a local home assistant (you can access all data via HA via vpn or HA cloud).

bjornenvall, 2024-06-02:
**thank you so mush for this!** close to plug n play program. m5stack s3 lite.

bjornenvall, 2024-06-02:
i have a daikin ac that I'm trying to do the same thing with. can't u make this too? I'm struggling with this: https://github.com/joshbenner/esphome-daikin-s21

gekkekoe, 2024-06-06:
Sorry don't have a unit, so unable to test, but it does look similar