gekkekoe/esphome-ecodan-hp #15: Esp Internal temperature - worng scale?

glowaa, 2024-05-24:
In wich scale is ESP internal temperature?
My WROOM32 show 75-83C, but when i touch it, it is only slightly warm.
I think that is Farenheit value not Celsius.

gekkekoe, 2024-05-24:
It's celcius for me for sure.  also according to the doc https://esphome.io/components/sensor/internal_temperature.html

glowaa, 2024-05-24:
Note
Some ESP32 variants return a large amount of invalid temperature values, including 53.3°C which equates to a raw value of 128. Invalid measurements are ignored by this component.

Raw value is in Farenheit.

gekkekoe, 2024-05-24:
I use the standard component and don't do anything with the values. it states that invalid values will be ignored. if's its fahrenheit for your board, then you probably define a filter in the yaml to do the conversion. But this seems to be an issue with the component in combination with your esp32 board. I've used couple of boards, and all of them are in celcius. So probably best to report it with esphome?

gekkekoe, 2024-05-24:
looking at https://www.esphome.io/api/internal__temperature_8cpp_source, you progably need to find the correct board definition for your esp32. 

your board probably needs this define
```
#if defined(USE_ESP32_VARIANT_ESP32)
   uint8_t [raw](https://www.esphome.io/api/bl0939_8h#ae24fc98981731522f51defd5d5e7fc95) = [temprature_sens_read](https://www.esphome.io/api/internal__temperature_8cpp#a2188b2e5ed72e93782e0a6d59111a180)();
   ESP_LOGV(TAG, "Raw temperature value: %d", raw);
   temperature = (raw - 32) / 1.8f;
   success = (raw != 128);
```

glowaa, 2024-05-27:
I have add to yaml

sensor:
  - platform: internal_temperature
    name: "ESP Internal Temperature"
    unit_of_measurement: "°F"

and now have right Celsius value

gekkekoe, 2024-05-27:
great

closed 2024-05-27 by gekkekoe