gekkekoe/esphome-ecodan-hp #16: DHW current temp not updating

glowaa, 2024-06-01:
After you added climate control, DHW current temp not updating.
Also heatpump frequently update a lot of statuses.

gekkekoe, 2024-06-01:
please use 
source: github://gekkekoe/esphome-ecodan-hp@8a8a51de65731c120eb0168c350ba484ddecc54d
in the meantime

glowaa, 2024-06-01:
Cant change source

INFO ESPHome 2024.5.4
INFO Reading configuration /config/esphome/pompa-ciepla.yaml...
INFO Updating https://github.com/gekkekoe/esphome-ecodan-hp@main
INFO Updating https://github.com/gekkekoe/esphome-ecodan-hp.git@8a8a51d
Failed config

external_components: [source /config/esphome/pompa-ciepla.yaml:41]
  - 
    couldn't find remote ref 8a8a51d.
    source: github://gekkekoe/esphome-ecodan-hp@8a8a51d
    components: 
      - ecodan
    refresh: always

gekkekoe, 2024-06-01:
8a8a51de65731c120eb0168c350ba484ddecc54d  use full hash

glowaa, 2024-06-11:
Version 8a8a51de65731c120eb0168c350ba484ddecc54d is working stable without any problems for over a week

gekkekoe, 2024-06-11:
okay thanks will merge it end of this week

gekkekoe, 2024-06-14:
I've replaced the standard climate controls with a custom climate control. Try the latest and report back. If this still gives issues, then you can always use the previous working version.

glowaa, 2024-06-15:
[image]

After today climate actions heatpump is running even the temperature is bigger than on room sensor

gekkekoe, 2024-06-15:
climates don't turn on heating. it's only displaying. from your logging, I see that space heating was activated. that's why it's showing heating.

you have room temp target set at 24 degrees, the heat pump tries to heat to that temp.

glowaa, 2024-06-16:
I have set target to 20 degrees and is 24 is currently on room sensor.

gekkekoe, 2024-06-16:
that is weird. dhw indication had a bug, fixed.
but normal climate should work properly.

the only set it does is setting the target temp depending on you operating mode. did you by change switched the modes in the thermostats? default it switches to cool flow for cooling and compensation for heating.

but I do see space heating activated, so something is requesting heat (thermostat can't do that, unless you set target below room temp (for room modes))

is it still giving issues? please post the operating modes, room temp sensor values, INx thermostats values

glowaa, 2024-06-16:
I have update to the newest version.

[image]

Now DHW seems be ok, but heating not.
After update heatpump change automaticly to Heat Compensation Curve and heating start, after change to Heat Target Temperature still is heating and waterpump is active.

gekkekoe, 2024-06-16:
I just pushed an update, please test that. It's a bit difficult since I cannot reproduce it here.
If it still gives issues, please revert to https://github.com/gekkekoe/esphome-ecodan-hp/commit/8a8a51de65731c120eb0168c350ba484ddecc54d

glowaa, 2024-06-16:
After update sill heating and waterpump on

[image]

8a8a51d working values

3-way Valve	OFF	
Boiler Flow Temp	25.0 °C	
Boiler Return Temp	25.0 °C	
Booster Heater	OFF	
Compressor Frequency	0.00 Hz	
Controller Version	FTC6	
Cool COP	0.00	
Cool Consumed	0.00 kWh	
Cool Delivered	0.00 kWh	
DHW	Off	
DHW COP	2.91	
DHW Consumed	0.88 kWh	
DHW Current Temp	46.5 °C	
DHW Delivered	2.56 kWh	
DHW Forced	OFF	
DHW Max Temp Drop	8.0 °C	
DHW Mode	Eco	
Eco
DHW Prohibit	OFF	
DHW Secondary Temp	25.0 °C	
DHW Setpoint	47.0 °C	
40.0 
 60.0
DHW Setpoint Value	47.0 °C	
Defrost	OFF	
ESP IP	192.168.203.123	
ESP Internal Temperature	76.7 °F	
ESP SSID	wifi	
ESP Uptime	3808 s	
ESP Version	2024.5.5	
Estimated Output Power	0.000 kW	
Feed Temp	29.0 °C	
Flow Rate	0 L/min	
Force DHW	OFF	
Heat Source	Heatpump	
Heating COP	8.83	
Heating Consumed	0.06 kWh	
Heating Delivered	0.53 kWh	
Heatpump	ON	
Holiday Mode	OFF	
Holiday Mode	OFF	
Immersion Heater	OFF	
In1 Thermostat	OFF	
In5 Outdoor Thermostat	OFF	
In6 Thermostat 2	OFF	
Legionella Prevention Temp	65.0 °C	
Operating Runtime	3381 h	
Operating mode	Heat Target Temperature	
Heat Target Temperature
Operation Mode	Off	
Output Power	0.000 kW	
Outside Temp	19 °C	
Refrigerant Condensing Temp	26.0 °C	
Refrigerant Liquid Temp	27.5 °C	
Return Temp	27.0 °C	
Water Pump Status	OFF	
WiFi Signal %	68 %	
WiFi Signal dB	-66 dBm	
Zone 1 H/C Setpoint	20.0 °C	
5.0 
 60.0
Zone 1 H/C Setpoint Value	29.0 °C	
Zone 1 Prohibit Cooling Status	ON	
Zone 1 Prohibit Heating	OFF	
Zone 1 Room Setpoint	22.5 °C	
8.0 
 28.0
Zone 1 Room Setpoint Value	22.5 °C	
Zone 1 Room Temp	24.5 °C	

AND values after latest update

3-way Valve	OFF	
Boiler Flow Temp	25.0 °C	
Boiler Return Temp	25.0 °C	
Booster Heater	OFF	
Compressor Frequency	0.00 Hz	
Controller Version	FTC6	
Cool COP	0.00	
Cool Consumed	0.00 kWh	
Cool Delivered	0.00 kWh	
DHW	Off	
DHW COP	2.91	
DHW Climate	IDLE	Current: 46.5, Target: 47.0
5.0 
 60.0
Mode:

HEAT
DHW Consumed	0.88 kWh	
DHW Current Temp	46.5 °C	
DHW Delivered	2.56 kWh	
DHW Forced	OFF	
DHW Max Temp Drop	8.0 °C	
DHW Mode	Eco	
Eco
DHW Prohibit	OFF	
DHW Secondary Temp	25.0 °C	
DHW Setpoint Value	47.0 °C	
Defrost	OFF	
ESP IP	192.168.203.123	
ESP Internal Temperature	80.0 °F	
ESP SSID	wifi	
ESP Uptime	233 s	
ESP Version	2024.5.5	
Estimated Output Power	0.000 kW	
Feed Temp	22.5 °C	
Flow Rate	13 L/min	
Force DHW	OFF	
Heat Source	Heatpump	
Heating COP	8.83	
Heating Consumed	0.06 kWh	
Heating Delivered	0.53 kWh	
Heatpump	ON	
Holiday Mode	OFF	
Holiday Mode	OFF	
Immersion Heater	OFF	
In1 Thermostat	OFF	
In5 Outdoor Thermostat	OFF	
In6 Thermostat 2	OFF	
Legionella Prevention Temp	65.0 °C	
Operating Runtime	3381 h	
Operation Mode	Space Heating	
Output Power	0.000 kW	
Outside Temp	19 °C	
Refrigerant Condensing Temp	26.0 °C	
Refrigerant Liquid Temp	27.5 °C	
Return Temp	22.5 °C	
Selected operating mode	Heat Target Temperature	
Heat Target Temperature
Water Pump Status	ON	
WiFi Signal %	60 %	
WiFi Signal dB	-70 dBm	
Zone 1 Climate	HEATING	Current: 24.5, Target: 22.5
5.0 
 60.0
Mode:

HEAT
Zone 1 H/C Setpoint Value	23.0 °C	
Zone 1 Prohibit Cooling Status	ON	
Zone 1 Prohibit Heating	OFF	
Zone 1 Room Setpoint Value	22.5 °C	
Zone 1 Room Temp	24.5 °C

Thebasestar, 2024-06-16:
I do have climate on heat as well, if set to off it restore itsef to heat immediatly, pump stay off probably because i did set Heating prohibited in this season.
Got this from log
[16:25:32][W][select:111][serial_rx_task]: 'Modalità Funzionamento' - Option Heat Compensation Curve is not a valid option

Yesterday was ok

gekkekoe, 2024-06-16:
the climate shows the selected heating/cooling mode. So heating when you have selected 'heat flow/room temp/compensation curve'. The action indicate if it's actually busy. It will show Idle probably in your case. Once it's heating, you will see the action turn into heating.

You can configure the heating/cooling switch. see updated readme. But it only switches to that mode.
Regarding the error, I think it's an translation issue. check if options for mode select is okay ?

@glowaa I see space heating is active, but compressor is at 0hz. but pump is on. This is probably during startup. The selected mode is space heating, So would not expect it to be active. can you try to poweroff the unit (via circuit breaker) and turn it on again? Is it still giving issues?

glowaa, 2024-06-16:
Seems be working ok now.
I will wait what about actions from node-red automotion.

gekkekoe, 2024-06-16:
good, so the not updating is solved with these custom climates? Did you had any automations that uses the number sliders ? You should change that to the climate ones. I've temporary enabled the sliders, but will remove them once there are no complaints about the climates.

glowaa, 2024-06-17:
Automotions works, but ESP statuses is not stable

[image]

[image]

For now https://github.com/gekkekoe/esphome-ecodan-hp/commit/8a8a51de65731c120eb0168c350ba484ddecc54d is one without this issue.

gekkekoe, 2024-06-17:
hmm try to run https://github.com/gekkekoe/esphome-ecodan-hp/commit/8a8a51de65731c120eb0168c350ba484ddecc54d and monitor DHW mode. The climates should not affect that sensor. Could be that wifi is dropping. I made some small improvements, please test latest and report back. Also try https://github.com/gekkekoe/esphome-ecodan-hp/commit/8a8a51de65731c120eb0168c350ba484ddecc54d for a couple of hours and compare dhw mode.

glowaa, 2024-06-17:
On https://github.com/gekkekoe/esphome-ecodan-hp/commit/8a8a51de65731c120eb0168c350ba484ddecc54d whole week without this issue and wifi is not droping i have that checked on my AP.

Newest version not install.

INFO ESPHome 2024.5.5
INFO Reading configuration /config/esphome/pompa-ciepla.yaml...
INFO Updating https://github.com/gekkekoe/esphome-ecodan-hp/@main
WARNING Found 'id(ecodan_instance).set_hp_mode(static_cast<uint8_t>(esphome::ecodan::Status::HpMode::${default_heating_switch_mode}));' (see climate->1->heatpump_climate_z1->heating_switch_func) which looks like a substitution, but 'default_heating_switch_mode' was not declared
WARNING Found 'id(ecodan_instance).set_hp_mode(static_cast<uint8_t>(esphome::ecodan::Status::HpMode::${default_cooling_switch_mode}));' (see climate->1->heatpump_climate_z1->cooling_switch_func) which looks like a substitution, but 'default_cooling_switch_mode' was not declared
WARNING Found 'id(ecodan_instance).set_hp_mode(static_cast<uint8_t>(esphome::ecodan::Status::HpMode::${default_heating_switch_mode}));' (see climate->1->heatpump_climate_z1->heating_switch_func) which looks like a substitution, but 'default_heating_switch_mode' was not declared
WARNING Found 'id(ecodan_instance).set_hp_mode(static_cast<uint8_t>(esphome::ecodan::Status::HpMode::${default_cooling_switch_mode}));' (see climate->1->heatpump_climate_z1->cooling_switch_func) which looks like a substitution, but 'default_cooling_switch_mode' was not declared
INFO Updating https://github.com/gekkekoe/esphome-ecodan-hp.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing pompa-ciepla (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/pompa-ciepla/src/main.cpp.o
src/main.cpp: In lambda function:
src/main.cpp:501:138: error: '$' is not a member of 'esphome::ecodan::Status::HpMode'
   heatpump_climate_z1->set_heating_func([=](void){ id(ecodan_instance).set_hp_mode(static_cast<uint8_t>(esphome::ecodan::Status::HpMode::${default_heating_switch_mode})); });
                                                                                                                                          ^
src/main.cpp:501:139: error: expected ')' before '{' token
   heatpump_climate_z1->set_heating_func([=](void){ id(ecodan_instance).set_hp_mode(static_cast<uint8_t>(esphome::ecodan::Status::HpMode::${default_heating_switch_mode})); });
                                                                                                                                           ^
                                                                                                                                           )
src/main.cpp: In lambda function:
src/main.cpp:502:138: error: '$' is not a member of 'esphome::ecodan::Status::HpMode'
   heatpump_climate_z1->set_cooling_func([=](void){ id(ecodan_instance).set_hp_mode(static_cast<uint8_t>(esphome::ecodan::Status::HpMode::${default_cooling_switch_mode})); });
                                                                                                                                          ^
src/main.cpp:502:139: error: expected ')' before '{' token
   heatpump_climate_z1->set_cooling_func([=](void){ id(ecodan_instance).set_hp_mode(static_cast<uint8_t>(esphome::ecodan::Status::HpMode::${default_cooling_switch_mode})); });
                                                                                                                                           ^
                                                                                                                                           )
*** [.pioenvs/pompa-ciepla/src/main.cpp.o] Error 1
========================= [FAILED] Took 11.88 seconds =========================

gekkekoe, 2024-06-17:
configure cooling/heating action in ecodan-esphome.yaml (see readme, default cooling/heating switch needs to be defined)
add these lines https://github.com/gekkekoe/esphome-ecodan-hp/blob/050785cb841b0f7b7c852d27532ab3b0f526fa9e/ecodan-esphome.yaml#L4
you can configure the default cool/heat switch yourself this way.

glowaa, 2024-06-18:
Seems to be working ok now.

gekkekoe, 2024-06-18:
good, I think it's a timing issue, I turned of the climate, but latest version changes that. it's display only.

closed 2024-06-18 by gekkekoe