gekkekoe/esphome-ecodan-hp #19: sensor.ecodan_dhw_consumed And sensor.ecodan_heating_consumed

bjornenvall, 2024-06-12:
Hi, i want to use the 2 sensors in the energy tab. But it doesn’t seem to work because they are not only increasing. Do you have any idea how to fix this? 
[image]

bwduncan, 2024-06-12:
Can you use a utility meter with delta values = True?

gekkekoe, 2024-06-13:
the values are reported once a day. But to be honest they are far from accurate. It's a nice indication. I actually have a certified kwh meter and can compare the results over 24h.

But If you want to go with the ecodan numbers, then I think you can use a utility meter as @bwduncan suggest.
I did hear from ppl that when you connect a pulse kwh meter to the ecodan, the results becomes accurate (but you can read the pulse kwh meter directly...)