gekkekoe/esphome-ecodan-hp #24: Problem after update to lastest version (climate: fix current/target temp getters when in compensation mode)

glowaa, 2024-06-20:
Again with statuses "Became unavailable"

Pompa ciepla DHW Mode changed to Eco
10:00:41 AM - 5 seconds ago
Pompa ciepla DHW Mode became unavailable
10:00:39 AM - 7 seconds ago
Pompa ciepla DHW Mode changed to Eco
10:00:18 AM - 28 seconds ago
Pompa ciepla DHW Mode became unavailable
10:00:16 AM - 30 seconds ago
Pompa ciepla DHW Mode changed to Eco
9:59:11 AM - 2 minutes ago
Pompa ciepla DHW Mode became unavailable
9:59:09 AM - 2 minutes ago
Pompa ciepla DHW Mode changed to Eco
9:58:36 AM - 2 minutes ago
Pompa ciepla DHW Mode became unavailable
9:58:34 AM - 2 minutes ago
Pompa ciepla DHW Mode changed to Eco
9:56:56 AM - 4 minutes ago
Pompa ciepla DHW Mode became unavailable
9:56:54 AM - 4 minutes ago
Pompa ciepla DHW Mode changed to Eco
9:56:19 AM - 4 minutes ago
Pompa ciepla DHW Mode became unavailable
9:56:17 AM - 4 minutes ago
Pompa ciepla DHW Mode changed to Eco
9:55:18 AM - 5 minutes ago
Pompa ciepla DHW Mode became unavailable
9:55:16 AM - 5 minutes ago
Pompa ciepla DHW Mode changed to Eco
9:54:21 AM - 6 minutes ago
Pompa ciepla DHW Mode became unavailable
9:54:19 AM - 6 minutes ago

Pompa ciepla Zone 1 Climate changed to Heat
10:00:41 AM - 40 seconds ago
Pompa ciepla Zone 1 Climate became unavailable
10:00:39 AM - 41 seconds ago
Pompa ciepla Zone 1 Climate changed to Heat
10:00:18 AM - 1 minute ago
Pompa ciepla Zone 1 Climate became unavailable
10:00:16 AM - 1 minute ago
Pompa ciepla Zone 1 Climate changed to Heat
9:59:11 AM - 2 minutes ago
Pompa ciepla Zone 1 Climate became unavailable
9:59:09 AM - 2 minutes ago
Pompa ciepla Zone 1 Climate changed to Heat
9:58:36 AM - 3 minutes ago
Pompa ciepla Zone 1 Climate became unavailable
9:58:34 AM - 3 minutes ago
Pompa ciepla Zone 1 Climate changed to Heat
9:56:56 AM - 4 minutes ago
Pompa ciepla Zone 1 Climate became unavailable
9:56:54 AM - 4 minutes ago
Pompa ciepla Zone 1 Climate changed to Heat
9:56:19 AM - 5 minutes ago
Pompa ciepla Zone 1 Climate became unavailable
9:56:17 AM - 5 minutes ago
Pompa ciepla Zone 1 Climate changed to Heat
9:55:18 AM - 6 minutes ago
Pompa ciepla Zone 1 Climate became unavailable
9:55:16 AM - 6 minutes ago
Pompa ciepla Zone 1 Climate changed to Heat
9:54:21 AM - 7 minutes ago
Pompa ciepla Zone 1 Climate became unavailable
9:54:19 AM - 7 minutes ago

gekkekoe, 2024-06-20:
can you see in the logging if home assistant disconnects at those times? For me it was home assistant that was giving issues. After update, no more disconnects. Try updating to latest home assistant and home assistant OS if you are using those.

glowaa, 2024-06-20:
HA in in the lastest version ESPHome too.

I only change latest to https://github.com/gekkekoe/esphome-ecodan-hp/commit/050785cb841b0f7b7c852d27532ab3b0f526fa9e and it working again.

gekkekoe, 2024-06-20:
is the climate only becoming unavailable or other sensors as well? no issues locally here on both versions

glowaa, 2024-06-20:
on other sensors too

gekkekoe, 2024-06-20:
the latest change only affect climateas, but I had these issues too. Please update to latest esphome/home assistant. And do a full HA reboot (not the quick reload)

when you monitor esphome logs --device ip ecodan-esphome.yaml, you will see that home assistant disconnects at those times.
there seems to be an issues with one of the latest HA/ESPHome. Try and report back.

gekkekoe, 2024-06-20:
Esphome is not rebooting, you can see it in time uptime sensor. It's HA that disconnect from esphome for some reason.
I upgraded to esphome 2024.6.1 and see issue again. This is really strange.

I've reverted NAN return part. Try and see if it works

glowaa, 2024-06-20:
1 hour and so far is ok

gekkekoe, 2024-06-20:
I don't think it will solve it, does not make sense.

https://esphome.io/guides/faq.html#my-node-keeps-reconnecting-randomly

when all sensors become unavailable, then it had disconnect. seems to be a known issue.

glowaa, 2024-06-20:
3 hour and still working ok

gekkekoe, 2024-06-20:
I already saw 3 reconnects here

glowaa, 2024-06-21:
14 hours and rock solid :)

gekkekoe, 2024-06-21:
[image]
good that it works for you, but I'm still convinced that it's network related. The change should not affect a disconnect (only diff is returning NAN vs current temp in thermostat for unknown modes). If it was related to the change, then only the climates should be affected (code does not crash component).

glowaa, 2024-06-21:
I have HA locally (8 core cpu, 12gb ram and nvme disk) and do not change netwrok settings at all.
Only change firmware on ESP and some versions work normally and others disconnect frequently.

gekkekoe, 2024-06-24:
For me it's automatically removed by home assistant. But you can remove the integration and add it again. Then it will be cleaned as well.

closed 2024-06-24 by gekkekoe