gekkekoe/esphome-ecodan-hp #34: What are reporting consumed sensors?

rzulian, 2024-08-03:
What are consumed sensors reporting? It's not really clear. 
I though it was the daily consumed, but it doesn't look so. Moreover they are not changing at midnight, is this due to the time saving period?

 
[image]

gekkekoe, 2024-08-03:
Previous daily consumed kWh (heating/cool/DHW). 

it is reported around midnight. you probably have the time (summer time switch) set incorrectly.

btw, questions can be asked in the discussions section. will move this there eventually.

rzulian, 2024-08-03:
Do you is incorrect in the ecodan or in the component?
Do I need an utility meter to get it correctly, assuming the time is correct?

gekkekoe, 2024-08-03:
setting in ecodan that has to be manually switched each season...

for meters please read  https://github.com/gekkekoe/esphome-ecodan-hp/discussions/22