gekkekoe/esphome-ecodan-hp #44: Missing Energy Delivery/Consumption & COP

Masterwork, 2024-08-28:
So far everything is working fine but there are no consumption and delivery and as well the COP for heating and cooling is missing.
Could you help me with that?
[image]

F1p, 2024-08-28:
This data is published at midnight (FTC controller time) with figures of the previous day - It is not a live counter.

Masterwork, 2024-08-28:
Thanks a lot for the quick response!
I just setup the ESP this afternoon. You are fully right, the data was transferred at midnight.
[image]
Thank you very much!

closed 2024-08-28 by Masterwork