gekkekoe/esphome-ecodan-hp #46: Missing "selected operating mode" for zone2

wolterink, 2024-09-08:
Great project so far,

I'm trying to determine if there should be a option to set the operating mode for zone 2 independently or not.

I'm my config I can only set the operating mode for zone 1 and not 2. Changing the mode only change the setting in the controller for zone 1, zone2 stays as is. I'm able to change the mode on the controller itself and just not via the esp controller. 

I enabled the zone2.yaml and all other controls do show up for zone 2, just not the operating mode (flow temp, cool temp,weather compensation, etc)
I did have to change the code slightly to make it work with my esp32 ethernet module, but I only took the wifi coding out and left the rest in tact.
[image]

Any ideas to have the option enabled for zone 2?

gekkekoe, 2024-09-08:
how would that work ? don't think the ecodan can do different modes for zone 1 vs zone 2? 
I did check the procon manual, and it seems possible to set the "A/C Mode – Zone 2" independently from "A/C Mode – Zone 1". Sadly I don't have a procon anymore so cannot reverse engineer this cmd easily.

wolterink, 2024-09-08:
My intension is not to have them in different modes, either both in set flow temp or heat compensation. The problem is, if I change the mode for zone 1, zone 2 remains in its previous set mode. So now they will operate in different modes. I haven't seen what behaviour this causes and who wins with its settings. (unit still being installed).

[image]

If changing the operating mode setting causes both zone 1 and 2 to change to the new setting, I would be happy with that. As there is no need to have them set independently. 

Changing the unit to cooling mode, does change the settings for zone 2 to match the cooling mode (unable to have one set for cooling and the other to heating). Same when it's changed back to heating.

gekkekoe, 2024-09-08:
So If we always set Z1 and Z2 with the same operating mode that would work for you? Thinking of it, that probably a good idea (was suggested by @F1p).

gekkekoe, 2024-09-08:
I pushed the change, please rebuild and test and report if it works properly

F1p, 2024-09-08:
Interesting picture, i have 2 zones - but one selection option for mode, so i keep synced:

[image]
[image]

wolterink, 2024-09-08:
Syncing it would be perfect.

I did do a rebuild, but unfortunately it had no effect.

gekkekoe, 2024-09-08:
are you using a local version of the repo ?  and you did flash the new firmware right?

Cause I'm doing the same as https://github.com/F1p/Mitsubishi-CN105-Protocol-Decode/blob/b5d98bfbb02f09f4de870cf1e8b64746817f2f67/EcodanDecoder.cpp#L701

wolterink, 2024-09-08:
No luck, I can only see zone 1 change on the screen.

my config: only the base and esp32 are local files

external_components:
  - source: github://gekkekoe/esphome-ecodan-hp@main
    components: [ ecodan ]
    refresh: always

packages:
  base: !include ecodan/base.yaml
  esp32: !include ecodan/esp32.yaml
  
  remote_package:
    url: https://github.com/gekkekoe/esphome-ecodan-hp/
    ref: main
    refresh: always
    files: [ 
###            confs/base.yaml,        # required
###            confs/esp32s3.yaml,     # confs/esp32.yaml, for regular board
            confs/zone1.yaml,
            ## enable if you want to use zone 2
            confs/zone2.yaml,
            ## enable label language file
            confs/ecodan-labels-en.yaml,
            #confs/ecodan-labels-nl.yaml,
            #confs/ecodan-labels-it.yaml,
            #confs/ecodan-labels-fr.yaml,
            confs/server-control.yaml,
            #confs/debug.yaml,
           ]

gekkekoe, 2024-09-08:
please check if the new firmware is flashed by checking the build number/date in HA, device info

wolterink, 2024-09-08:
It does show the new version
Firmware: 2024.8.3 (Sep 8 2024, 09:57:51)

I also tried a local repo with the updated commands file. 
No effect I'm afraid

gekkekoe, 2024-09-08:
When my legionella is finished later today I will have a look. I can read back the z2 value probably. but this work for others, so strange why it does not for you. What FTC are you using?

wolterink, 2024-09-08:
It’s a new unit with FTC6,

Maybe it’s an internal setting that is different?

I have extra sensors installed to measure the flow temps for Z1 + Z2 (flow return). They had to configure it as it had a buffer tank (no buffer tank installed). Maybe that has something to do with it.
No a big deal breaker, I’m happy with the config so far. Would be nice to have them synced

gekkekoe, 2024-09-08:
Ah  you have the zone kit ? Z1 feed/return was added recently, to what thw 6,7,8,9  was it wired? I still needed confirmation for those

gekkekoe, 2024-09-08:
I manage to check it locally and this is what I see:
when I set to compensation curve, logging shows
Operating mode z1: 2, z2: 1
When I set to cool fow:
 Operating mode z1: 4, z2: 4
 
 modes:
```
            HEAT_ROOM_TEMP = 0,
            HEAT_FLOW_TEMP = 1,
            HEAT_COMPENSATION_CURVE = 2,
            COOL_ROOM_TEMP = 3,
            COOL_FLOW_TEMP = 4
```

Could it be that z2 is dependent on z1. you probably cannot set both to compensation curve? Can you try via your FTC ?
at least the cooling is in sync, but heating seems a bit weird

wolterink, 2024-09-08:
I can set both to compensation curve

[image]

It makes sense that if you only use one zone it need to be controlled by a mode. If one zone would be off then the other can still work. But then if both are on which will get priority? If I would have a temp control valve it would be possible to have both run independent of each other. zone 1 running at 40 degrees while zone 2, ufh is reduce to 30. So it makes sense the can be controlled independently. Could it be that in my setup (buffer tank with temp regulating valve) the zone is defined differently?

for the sensors 
[image]
[image]
THW6 -9 for the flow return sensors and THW10 is not connected and but fixed to 25

gekkekoe, 2024-09-08:
are you seeing those corresponding with the sensors in esphome ? (boiler feed/return and z1/z2 feed/return) ?

I can imagine that the heatpump for example does compensation curve, and some mixing for zone 2.

for example:
set point z1:  40c, 
z2: 35c

then it will mix some colder water to get 35c for z2. But it would still run 1 mode, but with different set points. In that scenario it would make sense that the slave can only "follow" the master via mixing. When one zone reach its set point, the heatpump still runs the same program with mixing.

Question remains, can both have the master role. 
I also don't have an answer for wiring z1/z2 to boiler temps, I've seen installers do this before.

gekkekoe, 2024-09-08:
I created a test branch
https://github.com/gekkekoe/esphome-ecodan-hp/tree/opmode_test

It allows both modes to be set independently, this is just for testing, because I'm still wondering how the priorities would work.
can you modify the yaml from
`source: github://gekkekoe/esphome-ecodan-hp@main`
to
`source: github://gekkekoe/esphome-ecodan-hp@opmode_test`

use all remote files, and enable zone 2. Try to see if you can set it via the esp independently with this branch and report back.
Also it will report operating modes via the logs. you can monitor via
`esphome logs --device ip ecodan-esphome.yaml`

Try change modes and see in red what it's reporting. Also the internal status text sensor is enabled, you should be able to read z1/z2 operating mode there as well.

gekkekoe, 2024-09-08:
@F1p noticed I needed to set correct mask for zone 2, so it works now as expected locally on my machine.
Please refresh and rebuild. Should be synced now

wolterink, 2024-09-08:
That did the trick, Thank you!!

When setting the mode, zone 1 changes and zone 2 syncs 1 sec after. So that works now.

F1p, 2024-09-08:
Great news, thanks for confirming

gekkekoe, 2024-09-08:
great, could you please confirm the 
boiler feed/return
z1 feed/return
z2 feed/return

temps ? So  boiler is probably reporting z1 and z2 should be okay?

wolterink, 2024-09-08:
So far I can see the temps are being reported. As it isn't a fully working unit yet, I can't monitor the temp for correct sense. But it look good to me.

[image]
[image]
[image]

gekkekoe, 2024-09-08:
thanks a lot, I will close this issue then

closed 2024-09-08 by gekkekoe