gekkekoe/esphome-ecodan-hp #50: "Software Diagnosis" icon shown in FTC6

Algram, 2024-09-13:
Hello,

I successfully configured your project. Before I was using https://github.com/rbroker/ecodan-ha-local.

One thing I am wondering is that in the FTC6 unit it permanently shows the "Software Diagnosis" icon now.
[image]

What does this mean and how can I deactivate this? With the `ecodan-ha-local` project this did not happen.

Thanks a lot for your work!

F1p, 2024-09-13:
It's not software diagnosis - it's Server Control Mode, allow you to control the prohibits on DHW, Heating & Cooling

Algram, 2024-09-13:
> It's not software diagnosis - it's Server Control Mode, allow you to control the prohibits on DHW, Heating & Cooling

Thanks! How can I disable that? If I remove out the `confs/server-control.yaml` entry from the config it does not seem to fix it.

F1p, 2024-09-13:
> > It's not software diagnosis - it's Server Control Mode, allow you to control the prohibits on DHW, Heating & Cooling
> 
> Thanks! How can I disable that? If I remove out the `confs/server-control.yaml` entry from the config it does not seem to fix it.
> 
> EDIT: Ah, figured it out: Comment in the server control, then disable the switch in Home Assistant, then remove it again. Thanks.
> 
> @F1p Could you explain what use cases there are for the server control? Why do people use it?

You beat me to it 👍 

Server control mode is helpful in a few ways - primary is performing an "eco" on-demand hot water run.
When you transition from DHW Prohibit "On" to "Off" - the Heat Pump will run an eco hot water boost, where "Heat Now/Force" runs the heat pump at full power, reducing efficiency. 

Secondly, prohibits respond much faster than adjusting setpoints on thermostats, so if you want to create automation to run at dynamic times - removing prohibits is easier than adjusting your controller schedule.

closed 2024-09-13 by Algram

gekkekoe, 2024-09-13:
you can also turn off the server control switch? no need to remove it