gekkekoe/esphome-ecodan-hp #51: Selected operating mode gets reset on every restart

Algram, 2024-09-13:
Currently the `Selected Operating mode` gets reset to whatever value was configured here on reboot:
[image]

Would it be possible to read the current operating mode from the heatpump and set that as the select value? Otherwise every reboot potentially resets the currently running heatpump.

F1p, 2024-09-13:
This is same for switches i think, when i saw it writing at startup back to "off" states

gekkekoe, 2024-09-13:
I reboot pretty often (flashing new firmware).
My operating mode does not get reset? It's restored from the saved state on the flash storage. I assume a reboot is an esp reboot ?

gekkekoe, 2024-09-16:
please confirm if it's really an issue. issue will be closed otherwise

bwduncan, 2024-09-16:
This happened to me yesterday. I power cycled the heat pump and then didn't realise it was in compensation mode all day. I kept setting the flow temp to 20, and it kept getting set back to 31.5 and I couldn't understand why until I got home and looked at the pump :see_no_evil: I haven't had any time to debug this though, sorry.

gekkekoe, 2024-09-16:
ah power cycles, not esp reboot....
I think I got an idea where it fails:
https://github.com/gekkekoe/esphome-ecodan-hp/blob/70b2601d329a28b1407823c38ef760ddffc7f3a7/components/ecodan/climate.cpp#L83

after power cycle, operating mode is probably not known and when the mode differs from the thermostat it tries to set the thermostat heating mode.

bwduncan, 2024-09-16:
Hmm. I just flashed the latest master and when it rebooted it's now in comp mode. I didn't power cycle the pump this time.

gekkekoe, 2024-09-16:
weird ?!  I am in cool flow mode, but when I flash it remains in cool flow
what mode were you in prior to the reboot ?

bwduncan, 2024-09-16:
Heat flow

Would logs help?

bwduncan, 2024-09-16:
Bam! Reproducible. It stayed in heat flow after the flash, then i did an "ESP Restart" from HA and it changed to comp curve after the reboot. 

[logs.txt](https://github.com/user-attachments/files/17012328/logs.txt)

bwduncan, 2024-09-16:
(and yes, i see the `default_heating_switch_mode` but it would be nice to get this bug fixed)

gekkekoe, 2024-09-16:
```
[12:52:05][D][sensor:094]: 'H/C status': Sending state 2.00000  with 0 decimals of accuracy
[12:52:05][D][select:115]: 'Selected operating mode' - Set selected option to: Heat Compensation Curve
[12:52:05][D][select:015]: 'Selected operating mode': Sending state Heat Compensation Curve (index 2)
[12:52:05][D][sensor:094]: 'DHW Setpoint Value': Sending state 50.00000 °C with 1 decimals of accuracy
[12:52:05][D][climate:396]: 'Zone 1 Climate' - Sending state:
[12:52:05][D][climate:399]:   Mode: HEAT
[12:52:05][D][climate:401]:   Action: IDLE
[12:52:05][D][climate:419]:   Current Temperature: 28.50°C
[12:52:05][D][climate:425]:   Target Temperature: 30.00°C
```
It seems the selection component did it. But the heating/cooling status is reporting 2 back from the heatpump, that's compensation curve. The climate seems okay, let me see If I can reproduce it locally

gekkekoe, 2024-09-16:
I think I can trigger it.
Its because of the climate state and the operating mode selection box that conflicts.
When you set a different mode via the selection box than the default available climate modes, you will get these issues.
It's because the climates sees a mode change (or mode was changed from its previous state) and tries to set the correct mode. 

steps to reproduce:
1) set hp in heatflow and leave defaults for climate heat/cool 
2) on restart, the climate action is Idle, but it sees that the mode is in heating. So it will issue a switch to heating. 
3) hp is now in compensation curve

Don't know if this can be fixed cleanly other than removing the heat/cool switch from the climates.

bwduncan, 2024-09-16:
That sounds familiar, I think the old-school melcloud integration had this conflict too.

If it's not fixable I think putting a big sign up saying "the default in the yaml is important, you should think about setting it" will be sufficient. Is making the climates less smart an issue for usability? As in, if it looks like it should be able change mode, but it can't, that's bad.

thanks for looking into this :smile:

gekkekoe, 2024-09-16:
not really, its just more convenient. You can always switch mode via the select component. Just no shortcut via climates anymore.
Issue is that climate can only do one heat/cool mode. Else I could have matched it. Going to think about it, if there's no solution, the switching will be disabled in climates. Icons will only show modes.

gekkekoe, 2024-09-16:
pushed https://github.com/gekkekoe/esphome-ecodan-hp/commit/4f6455607d6839555fd2f905ad155621b53ae3bc to disable the switching (for now) there's just no clean way to do it. Locally states are correctly restored (after couple of mins) on reboot.

Please re-test and report back.

bwduncan, 2024-09-16:
Sorry, I had to go to work. Flashed and "ESP restart"d and it looks ok. the heat pump is off at the moment though, maybe that makes a difference, so I'll test it again tomorrow. Thanks!

gekkekoe, 2024-09-17:
No worries, report back when you have time. Thanks for testing.

closed 2024-09-23 by gekkekoe