gekkekoe/esphome-ecodan-hp #57: pins override

tzapu, 2024-10-06:
hello,

this is probably a silly questions so feel free to close.

i have an atom lite non s3 version which has a different pinout and paltform to the s3 sample config but looks to have the same features. is there a way to specify just pin overrides and not have to copy the whole config and change in my main esphome file?
how would you do changes like that to support a new board? might be worth mentioning in the readme as well for newbs

thank you very much

gekkekoe, 2024-10-06:
hmm that might be a good idea, but with separate board definitions, you are able to do board specific stuff (ethernet for example).

For parameters we can abuse the substitutions like we do with translations. You generally want to keep those defined in the main user file. But that would imply that all board will have the same pin layout. Not ideal.

Normally if you have a custom setting, do a `git pull` and edit the main yaml file to enable local files.
This is done by editing the `esphome-ecodan.yaml` like this:
```yaml
external_components:
- source:
    type: local
    path: components
  components: [ ecodan ]

packages:
 base: !include confs/base.yaml # 
 esp32: !include confs/esp32s3.yaml # esp32.yaml for regular board
 zone1: !include confs/zone1.yaml
# disable if you don't want to use zone 2
# zone2: !include confs/zone2.yaml
# change language labels to -en for English or -nl for Dutch
# substitutions: !include confs/ecodan-labels-nl.yaml
 substitutions: !include confs/ecodan-labels-en.yaml
 svc: !include confs/server-control.yaml
 dbg: !include confs/debug.yaml
 wifi: !include confs/wifi.yaml

# external_components:
#   - source: github://gekkekoe/esphome-ecodan-hp@main
#     components: [ ecodan ]
#     refresh: always

# packages:
#   remote_package:
#     url: https://github.com/gekkekoe/esphome-ecodan-hp/
#     ref: main
#     refresh: always
#     files: [ 
#             confs/base.yaml,        # required
#             confs/esp32s3.yaml,     # confs/esp32.yaml, for regular board
#             confs/zone1.yaml,
#             ## enable if you want to use zone 2
#             #confs/zone2.yaml,
#             ## enable label language file
#             confs/ecodan-labels-en.yaml,
#             #confs/ecodan-labels-nl.yaml,
#             #confs/ecodan-labels-it.yaml,
#             #confs/ecodan-labels-fr.yaml,
#             confs/server-control.yaml,
#             #confs/debug.yaml,
#             ## enable this to monitor WiFi status with ESP in-built LED
#             #confs/status_led.yaml,
#             confs/wifi.yaml
#            ]
```
This will disable the remote files and use local files. You can now add a local yaml file that you can refer to, and once it works, you can contribute it to my repo. Next time you can refer to that online file. And other users can also benefit from it with the same hardware.

tzapu, 2024-10-06:
much appreciated, gives me all i need to go forth