gekkekoe/esphome-ecodan-hp #67: Is there a way to manually operate the 3-way valve and pump1/2?

mharizanov, 2024-10-20:
Hello, as per the title: I am looking to run a de-stratification cycle on the DHW, that requires operating pump1 and the 3-way valve. I understand I can use set up SmartGrid to run the pump + additional really on the 3-way valve, but thought it would be more elegant to do it without these interventions. This function is available from the manual control on the FTC6 control pannel, so I thought maybe it could be activated through a serial code. Thanks

F1p, 2024-10-20:
Is de-stratification known in FTC6 as dry-up?

mharizanov, 2024-10-20:
> Is de-stratification known in FTC6 as dry-up?

No, that different. Basically my 300 l DHW tank has its electric heater element in the middle, as the heat exchanger is split around it. I use excess PV to heat it, but in reality only half of it benefits. I therefore need to run this de-stratification cycle, to equalize the whole volume's temperature

F1p, 2024-10-20:
Do you have a picture from the controller what options you use?

F1p, 2024-10-20:
Ah ok, the manual operation - required to power down

You can get the status of each pump, but not seen any possibility to force them via CN105.
You might look at another form of pump control external to the FTC

mharizanov, 2024-10-20:
I was afraid I would get this exact comment :) Alright, external manipulation it is.. Thanks

closed 2024-10-20 by mharizanov