gekkekoe/esphome-ecodan-hp #76: ESP WROOM32 - D1 mini

stanos4, 2024-11-21:
Hi, after quite some time I updated to the most recent version, from version probably somewhere in July (looks like 30. July was when I forked your repo). Things were working by then on my board, but updating to the most recent version, I can't connect to the heatpump anymore - the web server works, messages are sent to homeassistant, but the module won't connect to the heatpump.

my board reports as ESP32-D0WD-V3, (can be found as "D1 Mini ESP32" on aliexpress and similar)
using esp32 libraries in arduino works as expected (old implementation of ecodan bridge still works on the very same board)

ecodan-esphome.yaml:
always used confs/esp32.yaml - not working now

confs/esp32.yaml:
only changed the RX/TX pins to ones I use in arduino and that previously worked in your implementation, 3/1

do you perhaps have any suggestions what could be wrong? I also tried to go back in time with "git reset --hard CHECKSUM", but the code would not compile anymore, complaining about (default ecodan-esphome.yaml)

```
Failed config

ecodan: [source ecodan-esphome.yaml:87]
  id: ecodan_instance
  
  [rx_pin] is an invalid option for [ecodan]. Please check the indentation.
  rx_pin: GPIO2
  
  [tx_pin] is an invalid option for [ecodan]. Please check the indentation.
  tx_pin: GPIO1

```

gekkekoe, 2024-11-21:
sounds like old yaml. download latest and swap pins, should work

stanos4, 2024-11-21:
unfortunately no success - I just clean cloned your repo directly to a new folder, only added my secrets.yaml, and changed from esp32s3.yaml to esp32.yaml and to GPIO03. Still the same

do you maybe have a suggestion on how to debug? I'm a complete noob in esphome. I'm happy that I learned some way to debug with Arduino (and telnet), and then you made this very nice implementation :)

gekkekoe, 2024-11-21:
you need to use the new esp32.yaml if it's a non S3 esp

stanos4, 2024-11-21:
are you referring to use the confs/esp32.yaml? I'm using the most recent one from your repo, 1e415fe.

stanos4, 2024-11-21:
I get these debug messages:

```
[ecodan.component:261] Attempt to tx CONNECT_CMD!
[ecodan.component:112] No reply received from the heatpump in the last 2 minutes, going to reconnect...
```

trying to inspect the signal with a standard voltmeter - the voltage drop on TX connect attempt is much smaller and/or shorter than what I have with the other working code (~2.5 vs >3V). No voltage change is indeed observed on RX pin at all.

(Pins are assigned correctly)

gekkekoe, 2024-11-22:
try commenting out the pullup

gekkekoe, 2024-11-22:
you also used the latest ecodan-esphome.yaml right?

gekkekoe, 2024-11-22:
please post wire schema and relevant yaml config parts. Is you cable correct? Just try to swap rx/tx pin in software (just in case)

stanos4, 2024-11-22:
wire schema I'll have to draw, can try that bit later. Conceptually, I use logic level converter 3.3v-5v ("3.3V-5V 4Channels Logic Level Converter Bi-directional Switching Module CYT1076") connected on RX/TX for signal and VCC/GND for power. The very same connection worked with your old implementation well.

ecodan-esphome.yaml

```
substitutions:
  name: ecodan-heatpump
  friendlyName: Ecodan heatpump

esphome:
  name: ${name}
  friendly_name: ${friendlyName}

ota:
  - platform: esphome
#  password: !secret heatpump_ota_password

api:
  reboot_timeout: 0s
#  encryption:
#    key: !secret heatpump_encryption_key

# external_components:
# - source:
#     type: local
#     path: components
#   components: [ ecodan ]

# packages:
#  base: !include confs/base.yaml # 
#  esp32: !include confs/esp32s3-heishamon.yaml # esp32.yaml for regular board
#  zone1: !include confs/zone1.yaml
# # disable if you don't want to use zone 2
# # zone2: !include confs/zone2.yaml
# # change language labels to -en for English or -nl for Dutch
# # substitutions: !include confs/ecodan-labels-nl.yaml
#  substitutions: !include confs/ecodan-labels-en.yaml
#  svc: !include confs/server-control.yaml
#  dbg: !include confs/debug.yaml
#  wifi: !include confs/wifi.yaml

external_components:
  - source: github://gekkekoe/esphome-ecodan-hp@main
    components: [ ecodan ]
    refresh: always

packages:
  remote_package:
    url: https://github.com/gekkekoe/esphome-ecodan-hp/
    ref: main
    refresh: always
    files: [ 
            confs/base.yaml,        # required
            #confs/esp32s3.yaml,     # confs/esp32.yaml, for regular board
            confs/esp32.yaml,
            confs/zone1.yaml,
            ## enable if you want to use zone 2
            #confs/zone2.yaml,
            ## enable label language file
            confs/ecodan-labels-en.yaml,
            #confs/ecodan-labels-nl.yaml,
            #confs/ecodan-labels-it.yaml,
            #confs/ecodan-labels-fr.yaml,
            confs/server-control.yaml,
            confs/debug.yaml,
            ## enable this to monitor WiFi status with ESP in-built LED
            #confs/status_led.yaml,
            ## enable this to monitor status with custom led colors, uses https://github.com/esphome/esphome/pull/5814
            #confs/status_led_rgb.yaml,
            confs/wifi.yaml
           ]

# enable logger
logger:
  baud_rate: 0
  logs:
    esp32.preferences: DEBUG
    sensor: DEBUG
    text_sensor: DEBUG
    binary_sensor: DEBUG
    switch: DEBUG
    button: DEBUG
    number: DEBUG
    climate: DEBUG
    component: DEBUG
    ecodan.component: DEBUG

# Enable Web server.
web_server:
  port: 80
  #ota: true

ecodan:
  id: ecodan_instance
  uart_id: uart_ecodan

```

confs/esp32.yaml

```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended
    # Custom sdkconfig options
    sdkconfig_options:
      COMPILER_OPTIMIZATION_SIZE: y

uart:
  id: uart_ecodan
  rx_pin:
    number: 1
    mode:
      input: true
      pullup: true
  tx_pin:
    number: 3
    mode:
      output: true
  baud_rate: 2400
  parity: EVEN
  stop_bits: 1
  # debug:
  #   direction: BOTH
  #   dummy_receiver: false
  #   after:
  #     delimiter: "\n"
  #   sequence:
  #     - lambda: UARTDebug::log_hex(direction, bytes, ' ');

sensor:
  - platform: internal_temperature
    name: ${internal_esp_temperature}
    entity_category: diagnostic

```

in my working old approach, I used SwSerial to set-up the communication (this is fork of https://github.com/m000c400/Mitsubishi-CN105-Protocol-Decode.git that I tried to improve):

```
#include "Ecodan.h"
ECODAN HeatPump;
#define HEATPUMP_STREAM SwSerial
#define SERIAL_BAUD 2400
#define SERIAL_CONFIG SWSERIAL_8E1
HEATPUMP_STREAM.begin(SERIAL_BAUD, SERIAL_CONFIG, 3, 1); // Rx, Tx
pinMode(3, INPUT_PULLUP);

HeatPump.SetStream(&HEATPUMP_STREAM);
```

stanos4, 2024-11-22:
also, I tried swapping RX/TX both in the code and physically, no change.

gekkekoe, 2024-11-22:
I am a bit puzzled, I used a generic esp32 for a while without any issues

stanos4, 2024-11-22:
I will later check if I have another esp32 at home. Indeed, I also don't understand what's going on - some simple program of blinking the 2 rx/tx  pins in esphome works well.

gekkekoe, 2024-11-22:
Did you get any warnings when flashing ?
Don't remember if the generic esp32 also needs the 
[code]
esphome:
  platformio_options:
    board_build.flash_mode: dio
[/code]
part for esp-idf. If u change it to arduino, does it work ?

gekkekoe, 2024-11-22:
de gpio pins, try to define them as:
```
uart:
  id: uart_ecodan
  rx_pin:
    number: GPIO3
    mode:
      input: true
      pullup: true
  tx_pin:
    number: GPIO1
    mode:
      output: true
  baud_rate: 2400
  parity: EVEN
  stop_bits: 1
```

Also some boards want rx/tx on gpio 17/18. But since you had it working before, this is not the case

stanos4, 2024-11-22:
nothing seems to be helping (still need to find/try another board).
The most annoying is, I can't go back in history and successfully compile an old version :( - do you have any clue why that could be? e.g. with 

```
git reset --hard 7d29b6a2f98d3412d3e1dd45f8937c79a5969ec9
```

gekkekoe, 2024-11-22:
you can change my repo in yaml.
change @main to the commit you want.

but try to find the difference, then you can run the latest

stanos4, 2024-11-22:
thx, getting somewhere - I can connect now again, with an old version, I will troubleshoot/check when it starts to break down.

gekkekoe, 2024-11-22:
yes let me know the config diffs, would love to track this one down.

stanos4, 2024-11-22:
things broke down here, 8bfcd85, it's the commit where you moved gpio & baud rate from ecodan-esphome.yaml to confs/BOARD.yaml boards files + I see some changes in e.g. ecodan.cpp, perhaps others. Will try to narrow down/play around a bit if I can find out what the root-cause is.

gekkekoe, 2024-11-22:
Ah we can rule out arduino framework, since that commit still uses arduino

gekkekoe, 2024-11-22:
aaaaaaaaaaaaah I got it.
you are referring to my remote repo, but you are having local changes. If you want to compile local changes, then 
- git pull.
- adjust local files.
- adjust yaml to use local files

```
external_components:
- source:
    type: local
    path: components
  components: [ ecodan ]

packages:
 base: !include confs/base.yaml # 
 esp32: !include confs/esp32s3-heishamon.yaml # esp32.yaml for regular board
 zone1: !include confs/zone1.yaml
# disable if you don't want to use zone 2
# zone2: !include confs/zone2.yaml
# change language labels to -en for English or -nl for Dutch
# substitutions: !include confs/ecodan-labels-nl.yaml
 substitutions: !include confs/ecodan-labels-en.yaml
 svc: !include confs/server-control.yaml
 dbg: !include confs/debug.yaml
 wifi: !include confs/wifi.yaml

# external_components:
#   - source: github://gekkekoe/esphome-ecodan-hp@main
#     components: [ ecodan ]
#     refresh: always

# packages:
#   remote_package:
#     url: https://github.com/gekkekoe/esphome-ecodan-hp/
#     ref: main
#     refresh: always
#     files: [ 
#             confs/base.yaml,        # required
#             confs/esp32s3.yaml,     # confs/esp32.yaml, for regular board
#             confs/zone1.yaml,m5stack-atom-lite-proxy
#             ## enable if you want to use zone 2
#             #confs/zone2.yaml,
#             ## enable label language file
#             confs/ecodan-labels-en.yaml,
#             #confs/ecodan-labels-nl.yaml,
#             #confs/ecodan-labels-it.yaml,
#             #confs/ecodan-labels-fr.yaml,
#             confs/server-control.yaml,
#             #confs/debug.yaml,
#             ## enable this to monitor WiFi status with ESP in-built LED
#             #confs/status_led.yaml,
#             ## enable this to monitor status with custom led colors, uses https://github.com/esphome/esphome/pull/5814
#             #confs/status_led_rgb.yaml,
#             confs/wifi.yaml
#            ]

```
We disable the remote files and use the local files (that contains your local changes). before it worked, because all the settings where in the main yaml.

stanos4, 2024-11-22:
lovely. Thank you a lot! and sorry for this confusion - as I say, I'm a complete noob in esphome, so now you gave me a good lesson :1st_place_medal:

closed 2024-11-22 by stanos4