gekkekoe/esphome-ecodan-hp #91: [Atom S3 Lite] firmware is reset to esphome default on reboot

m-baumgartner, 2024-12-10:
I have installed esphome via https://web.esphome.io/ and configured my wifi settings there too.

I then compiled the project with the default yaml, only activated zone 2. I also filled secrets.yaml.
I then used OTA update from the web ui to replace the firmware with the resulted firmware.bin file.

The OTA update seem to be doing fine, I see all the ecodan specific values in the webui and i'm able to adopt the device into my home assistant instance.

However, as soon as I reboot the device (either from web ui, or from home assistant or by unplugging/replugging usb-c connector) then the stock esphome appears. ecodan firmware seems gone.

I suspect a bootloader issue but i'm not too sure at this point.

here are the very first logs at boot : 

```
Logs
[13:47:50][I][logger:171]: Log initialized
[13:48:10][C][safe_modESP-ROM:esp32s3-20210327
[13:48:10]Build:Mar 27 2021
[13:48:10]rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
[13:48:10]Saved PC:0x420a4a52
[13:48:10]SPIWP:0xee
[13:48:10]mode:DIO, clock div:1
[13:48:10]load:0x3fce3808,len:0x39c
[13:48:10]load:0x403c9700,len:0x9bc
[13:48:10]load:0x403cc700,len:0x28dc
[13:48:10]entry 0x403c98c0
[13:48:10][I][logger:171]: Log initialized
[13:48:10][C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[13:48:10][D][esp32.preferences:114]: Saving 1 preferences to flash...
[13:48:10][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[13:48:10][I][app:029]: Running through setup()...
[13:48:10][C][esp32_ble:032]: Setting up BLE...
[13:48:10][C][wifi:048]: Setting up WiFi...
[13:48:10][C][wifi:061]: Starting WiFi...
[13:48:10][C][wifi:062]:   Local MAC: [redacted]
[13:48:10][D][wifi:074]: Loaded saved wifi settings: [redacted]
[13:48:10][D][wifi:482]: Starting scan...
[13:48:10][W][component:157]: Component wifi set Warning flag: scanning for networks
[13:48:10][D][esp32_ble:284]: Enabling BLE...
[13:48:11][D][esp32_ble_server:077]: BLE server setup successfully
[13:48:11][D][esp32_improv.component:082]: Creating Improv service
[13:48:13][D][wifi:497]: Found networks:
[13:48:13][I][wifi:541]: - '[redacted]' [redacted]▂▄▆█
[13:48:13][D][wifi:542]:     Channel: 6
[13:48:13][D][wifi:543]:     RSSI: -56 dB
[13:48:13][I][wifi:541]: - '[redacted]' [redacted]▂▄▆█
[13:48:13][D][wifi:542]:     Channel: 6
[13:48:13][D][wifi:543]:     RSSI: -70 dB
[13:48:13][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[13:48:13][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[13:48:13][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[13:48:13][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[13:48:13][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[13:48:13][D][wifi:546]: - [redacted] [redacted][D][wifi:546]: - [redacted] [redacted][D][wifi:546]: - [redacted] [redacted][D][wifi:546]: - [redacted] [redacted][I][wifi:313]: WiFi Connecting to '[redacted]'...
[13:48:15][I][wifi:617]: WiFi Connected!
[13:48:15][C][wifi:428]:   Local MAC: 48:CA:43:39:E0:E0
[13:48:15][C][wifi:433]:   SSID: [redacted]
[13:48:15][C][wifi:436]:   IP Address: 192.168.1.74
[13:48:15][C][wifi:440]:   BSSID: [redacted]
[13:48:15][C][wifi:441]:   Hostname: 'esphome-web-39e0e0'
[13:48:15][C][wifi:443]:   Signal strength: -57 dB ▂▄▆█
[13:48:15][C][wifi:447]:   Channel: 6
[13:48:15][C][wifi:448]:   Subnet: 255.255.255.0
[13:48:15][C][wifi:449]:   Gateway: 192.168.1.254
[13:48:15][C][wifi:450]:   DNS1: 192.168.1.254
[13:48:15][C][wifi:451]:   DNS2: 0.0.0.0
[13:48:15][D][wifi:626]: Disabling AP...
[13:48:15][C][web_server:100]: Setting up web server...
[13:48:15][C][api:026]: Setting up Home Assistant API server...
[13:48:15][I][app:062]: setup() finished successfully!
[13:48:15][W][component:170]: Component wifi cleared Warning flag
[13:48:15][W][component:157]: Component api set Warning flag: unspecified
[13:48:15][I][app:100]: ESPHome version 2024.11.1 compiled on Nov 25 2024, 08:41:12
[13:48:15][I][app:102]: Project esphome.web version 24.11.1
...
```

I'm guessing the `rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)` isn't a good sign but i'm not exactly sure how to solve this.

gekkekoe, 2024-12-10:
are you sure that the wifi credentials are correct ? If the wifi signal is bad, it will go to AP mode after a while.

m-baumgartner, 2024-12-10:
> are you sure that the wifi credentials are correct ?
yes

> If the wifi signal is bad, it will go to AP mode after a while.
the wifi AP is a couple of meters away, the signal strength is good

The logs do indicate that the wifi connection is fine. (i have just redacted away the personal details such as mac adresses and SSID).

I will add that before rebooting the device, there is no issue with the wifi connection to the device.

gekkekoe, 2024-12-10:
It goes into AP mode quite fast, so that is weird, it only does that when there are connectivity issues. Does your router show some additional log why the client might disconnect? 

Can you build the firmware locally and flash it via usb ? instruction are on the main page. Just to rule out something went wrong with the OTA

m-baumgartner, 2024-12-10:
that solved it. thank you.

so it would seem possible to conclude that the OTA process isn't working quite right in my specific configuration.
maybe it's because I installed stock esphome first.

closed 2024-12-10 by m-baumgartner

gekkekoe, 2024-12-10:
I always do OTA as described on the main page, never fails. I assume the gui stuff failed somewhere

DamijanD, 2024-12-10:
I also do it always via OTA. The only "problem" I had is that I couldn't get it done on Windows, so I use one Ubuntu VM for this.

m-baumgartner, 2024-12-10:
i've tried the command-line based OTA after an initial instal via https://web.esphome.io/ and got the same disfunctionnal results.
so my semi-informed intuition is that this stems from the way the initial flashing is done rather than how the OTA update is done.