gekkekoe/esphome-ecodan-hp #93: Wrong outside temperature reported

achillebnt, 2024-12-20:
Hi, first i want to thank you a lot for this project, for me is very helpful, later i offer you a coffe 😉 .

I found a issue about the outside temperature: according to decode documentation protocol the outside temperature is calculeted with formula:  Outside Temp /2 - 40

In [response.cpp](https://github.com/gekkekoe/esphome-ecodan-hp/blob/main/components/ecodan/response.cpp) you use [get_float8](https://github.com/gekkekoe/esphome-ecodan-hp/blob/81c1912f16b69781e6a15d203205be452543d6c9/components/ecodan/proto.h#L366)
https://github.com/gekkekoe/esphome-ecodan-hp/blob/81c1912f16b69781e6a15d203205be452543d6c9/components/ecodan/response.cpp#L117
but the calculation in this function is reversed.
I notice in proto.h the next function [get_float_v2](https://github.com/gekkekoe/esphome-ecodan-hp/blob/81c1912f16b69781e6a15d203205be452543d6c9/components/ecodan/proto.h#L373) is the right one, i tested and the temperature is reported right.

```c++
// Used for most single-byte floating point values
float get_float8(size_t index)
{
    float value = payload()[index];
    return (value / 2) - 40.0f;
}

// Used for DHW temperature dropand  min/max SH flow temperature threshold
float get_float8_v2(size_t index)
{
    float value = payload()[index];
    return (value - 40.0f) / 2;
}
``` 

Here in HA before:
[image]
here after:
[image]

F1p, 2024-12-20:
At 4degC outdoor i get a HEX of 0x58, you seem to be getting 0x30

[image]

Formula verified in procon

Can you give us more information on your Indoor Unit & FTC version?

gekkekoe, 2024-12-20:
So I'm pretty sure the outdoor temp reading is correct (like @F1p procon verified and used by a lot of ppl), are you looking at the correct one ? Might be a translation error

achillebnt, 2024-12-20:
In HA is **sensor.ecodan_heatpump_t_esterna** it refer to italian translation:  **outside_temp: T. Esterna**

Log:
```
[13:29:36][D][uart_debug:114]: >>> FC 42 02 7A 10 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 27
[13:29:36][V][sensor:043]: 'Z1 T. Ambiente': Received new state 19.500000
[13:29:36][D][sensor:094]: 'Z1 T. Ambiente': Sending state 19.50000 °C with 1 decimals of accuracy
[13:29:36][I][ecodan.component:018]: Could not publish state of sensor 'z2_room_temp' with value: '0.000000'
[13:29:36][V][sensor:043]: 'T. Esterna': Received new state 6.000000
[13:29:36][D][sensor:094]: 'T. Esterna': Sending state 6.00000 °C with 0 decimals of accuracy
[13:29:36][V][sensor:043]: 'T. Mandata Liq. Refr.': Received new state 19.000000
[13:29:36][D][sensor:094]: 'T. Mandata Liq. Refr.': Sending state 19.00000 °C with 1 decimals of accuracy
[13:29:36][V][sensor:043]: 'T. Cond. Liq. Refr.': Received new state 15.500000
[13:29:36][D][sensor:094]: 'T. Cond. Liq. Refr.': Sending state 15.50000 °C with 1 decimals of accuracy
[13:29:36][W][component:237]: Component ecodan took a long time for an operation (76 ms).
[13:29:36][W][component:238]: Components should block for at most 30 ms.
[13:29:36][D][climate:396]: 'Z1 temperatura' - Sending state:
[13:29:36][D][climate:399]:   Mode: OFF
[13:29:36][D][climate:401]:   Action: IDLE
[13:29:36][D][climate:419]:   Current Temperature: 19.50°C
[13:29:36][D][climate:425]:   Target Temperature: 20.00°C
[13:29:36][D][uart_debug:114]: <<< FC 62 02 7A 10 0B 07 9E F0 C4 F0 C4 0B 07 6C 6F 34 00 00 00 00 D9
``` 

I see the GetRequest  **0x42** packet type **0x0b** (Zone 1 & 2 and Outside) and response **0x62** - **0x0b**

| 0 | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11 | 12 | 13 | 14 | 15 | 16
| -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
| 0x0b | Z1T | Z1T | Z2T | Z2T | ? | ? |   | RT | RT | CT | O |   |   |   |   |  

</markdown-accessiblity-table>
<ul dir="auto">
<li>Z1T : Zone1 Temperature * 100</li>
<li>Z2T : Zone2 Temperature * 100</li>
<li>RT : Refrigerant Temperature * 100 (Where TH2 is installed)</li>
<li>CT : Condensing Temperature /2 - 40</li>
<li>O : Outside Temp  /2 - 40</li>
</ul>

If i read it correctly the  OT (11) is 0x34 in decimal 52 so:
(52 / 2) - 40 = -14°
(52 - 40) / 2 = 6 °

Now -14° is impossible and 6° it's actually the outside temperature, i notice if you go to original repo by _m000c400_ in that [section](https://github.com/m000c400/Mitsubishi-CN105-Protocol-Decode?tab=readme-ov-file#0x0b---zone-1--2-and-outside-temperature) outside temperature is **O : Outside Temp +40 x 2**

Maybe the right formula is (OT-40)/2?

gekkekoe, 2024-12-20:
We are pretty confident about outside temp, so it must be something else.

- Can you check thermistors on the controller?
- Verify that installer put it at the correct socket
- is your sensor broken?
- what ftc and software are you using?

achillebnt, 2024-12-20:
@F1p  TH7 is in fact a negative value, it refer to out sensor?

[image]

@gekkekoe 
- How can i check it? With a multi meter check the resistance?
- I see the schematics and i find where is located i check it asap
- I suppose it can be broken
- In HA i see FTC5 v. 15.00

For completion i have these machines:
Internal: EHSC-VM2CR2 
External: PUHZ-SW120HA)

F1p, 2024-12-20:
@achillebnt TH7 is the outdoor sensor, it is therefore matching with home assistant at -16 and the formula is correct.

It looks like the sensor is faulty.
It is located on the back of the outside unit, in the top left

gekkekoe, 2024-12-20:
my guess is that you installer used the wrong thermistor for the outside temp. 
Can you find th7 and make a picture or write down the probe part number? Then you can check if they used the correct one.

Th7 is outside temp sensor. If you still have warranty, ask the installer why the outside temp is measuring Siberia temps...

DamijanD, 2024-12-21:
But isn't this sensor preinstalled on the outside unit?

gekkekoe, 2024-12-21:
> But isn't this sensor preinstalled on the outside unit?

yeah, so probably faulty sensor. Or could have been swapped during repair. It's a bit too much of a coincidence that it has the characteristic of the other sensor type.

achillebnt, 2024-12-21:
I found the connector it's a preinstalled wire, the red one in the picture
[image]
Following the red wire i found the outside sensor
[image]
I can't figure out how to open the white case so i didn't read sensor code (if it is present).

@gekkekoe how can i check it? It's running information in service menu? It return:
005 - 6
007 - 3
009 - 2

th6+7 it's not swapped and th3 is separated, you can see it in photo
[image]

At this point it's quite sure it's faulty sensor, i search for code you mentioned but i can't find nothing. Searching on eBay i found a similar sensor but code is different R01 E68 202
[image]

gekkekoe, 2024-12-21:
try S70E60202, I got a bunch of hits. For example https://www.gtservicegroup.com/45608-mitsubishi-electric-europe-b.v-sonda-temperatura-ambiente-thermistor-ambient,2-phase-p-mitsubishi.html

So th3 and th6 are close, so then they did not swapped it, the connector can only plug in one way, so probably faulty sensor. Do you still hae warranty ? else order part and replace the sensor. Was it always like this? I reckon the heatpump takes into account to outside temp to operate.

gekkekoe, 2024-12-21:
I think your unit should be PUHZ-SW120VHA right ? the PUHZ-SW120HA does not seem to exists.
Then the partnumber is S70E31202, Please do check the correct model.

achillebnt, 2024-12-22:
> I think your unit should be PUHZ-SW120VHA right ? the PUHZ-SW120HA does not seem to exist. Then the partnumber is S70E31202, Please do check the correct model.

You right, my bad, i misspelled it 😅
Service reference in label is **PUHZ-SW120vHAR4.UK** and yes part is **S70 E31 202 THERMISTOR (AMBIENT, 2-PHASE PIPE)**
 
> So th3 and th6 are close, so then they did not swapped it, the connector can only plug in one way, so probably faulty sensor. Do you still have warranty ? else order part and replace the sensor. Was it always like this? I reckon the heatpump takes into account to outside temp to operate.

I think so too, warranty was exired it's 2017 installation and checking the installation report from thecnician outside temperature was reported correctly (11°) and i remember he read the sensor value in the service menu.
I suppose the sensor read the ouside temperature to estimate if work in combo with internal booster to help the outside pump but now that function is not enabled.
However all is working fine, i don't now how much time is in this state, i casually discovered it after see the strange value on HA and 
now i surely replace sensors.

My apologiases for the wrong issue i reported but coincidences led me in wrong direction, it's not easy find and understand how this work (it's a long story).

**I really want to thank you for all super support!**

_By the way i'm using a ESP32- WROOM-32 with rx_pin: GPIO16 tx_pin: GPIO17 and direct wiring whit no resistence and still work fine._

[image]

gekkekoe, 2024-12-22:
no problem good luck replacing sensor, glad we could help. moving this to discussions