gekkekoe/esphome-ecodan-hp #101: Invalid encryption key

rzulian, 2025-01-02:
I'm getting a lot of this kind of errors:
2025-01-02 13:49:45.010 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "aioesphomeapi/_frame_helper/noise.py", line 205, in aioesphomeapi._frame_helper.noise.APINoiseFrameHelper.data_received
  File "aioesphomeapi/_frame_helper/noise.py", line 355, in aioesphomeapi._frame_helper.noise.APINoiseFrameHelper._handle_frame
  File "aioesphomeapi/_frame_helper/noise.py", line 108, in aioesphomeapi._frame_helper.noise.DecryptCipher.decrypt
cryptography.exceptions.InvalidTag
2025-01-02 13:49:45.014 WARNING (MainThread) [aioesphomeapi.connection] ecodan-heatpump @ 192.168.2.150: Connection error occurred: ecodan-heatpump @ 192.168.2.150: Invalid encryption key: received_name=ecodan-heatpump
2025-01-02 13:49:45.015 INFO (MainThread) [aioesphomeapi.reconnect_logic] Processing unexpected disconnect from ESPHome API for ecodan-heatpump @ 192.168.2.150
2025-01-02 13:49:45.056 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to ecodan-heatpump @ 192.168.2.150 in 0.023s
2025-01-02 13:49:45.140 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successful handshake with ecodan-heatpump @ 192.168.2.150 in 0.084s
2025-01-02 13:5

What's wrong?

rzulian, 2025-01-02:
Not sure if I'm mixing problems, but I've tried to update the firmware to the latest version and I'm getting this 
INFO Reading configuration /config/esphome/ecodan.yaml...
INFO Updating https://github.com/gekkekoe/esphome-ecodan-hp/@main
INFO Updating https://github.com/gekkekoe/esphome-ecodan-hp.git@main
Failed config

esp32: None
  board: esp32dev
  framework: 
    type: arduino
    version: recommended
    
    [sdkconfig_options] is an invalid option for [framework]. Please check the indentation.
    sdkconfig_options: 
      COMPILER_OPTIMIZATION_SIZE: y

I'll check the encryption key, thanks!

rzulian, 2025-01-02:
I've moved HA from a VM to a RPi last week. Probably Something went wrong during the restore process.
Now I have the ESP device running and connected to the Ecodan, and I can enter the web ui, but I cannot connect to it from the ESP add on in HA due the the error above, but I'm receiving data in HA :(
The yaml file the I can see n the add-on  has two major differences from your template:
``` 
esp32:
  board: esp32dev
  framework:
    type: arduino

ecodan:
  id: ecodan_instance
  rx_pin: GPIO3
  tx_pin: GPIO1

```
but I'm not sure this has been the one that I used in July to update the firmware.
What should I do?

gekkekoe, 2025-01-02:
most easiest is top restore your old encryption key in HA. if not possible, configure and flash new firmware with correct keys

Use lastest template since a lot was added since july