gekkekoe/esphome-ecodan-hp #103: ESP8266 does not work

Bokenbergen, 2025-01-03:
Hi,

I'm able to flash the ESP8266 but it does not boot. I can see the following in the serial logs:

`[12:46:57]Unhandled C++ exception: OOM
[12:46:57]
[12:46:57]>>>stack>>>
[12:46:57]
[12:46:57]ctx: cont
[12:46:57]sp: 3ffffc00 end: 3fffffd0 offset: 0010
[12:46:57]3ffffc10:  00000000 3ffffc90 00000074 40235bcc  
[12:46:57]3ffffc20:  3fff8d10 3fffa107 3fffb7fc 40204026  
[12:46:57]3ffffc30:  00000003 3fffb955 00000000 00000074  
[12:46:57]3ffffc40:  ffffffff 00000000 00000001 3fffb7b4  
[12:46:57]3ffffc50:  00000001 00000002 3ffffc98 40204199  
[12:46:57]3ffffc60:  3fffb7fc 00000000 00000010 40203cb8  
[12:46:57]3ffffc70:  3fffb710 00000000 00000015 00000002  
[12:46:57]3ffffc80:  00000071 3fffb7b4 00000000 40204271  
[12:46:57]3ffffc90:  3fff9854 00000003 3fffb8e4 00000071  
[12:46:57]3ffffca0:  3fff9854 3fff9857 3fff9858 40205449  
[12:46:57]3ffffcb0:  0000000c 3fffb8e4 3fffb708 402063dc  
[12:46:57]3ffffcc0:  3ffffd20 3ffffcf0 000000a7 0000000c  
[12:46:57]3ffffcd0:  3fffb708 00000000 3fffb6fc 40201639  
[12:46:57]3ffffce0:  3ffffd20 3fffb708 3ffffd20 40206c80  
[12:46:57]3ffffcf0:  3fffb708 3fffb6fc 3ffffda8 4021e114  
[12:46:57]3ffffd00:  00000000 7972616e 3fffb6fc 3fff31b4  
[12:46:57]3ffffd10:  3fff8d10 3fffb6fc 3fff7604 40202ce1  
[12:46:57]3ffffd20:  4023bcec 3fffb814 00000017 00000017  
[12:46:57]3ffffd30:  3fffb404 000000ff 00000000 a7f35b40  
[12:46:57]3ffffd40:  3fffb85c 00000017 00000017 4024d49c  
[12:46:57]3ffffd50:  00000000 00000000 3fffbfac 00000026  
[12:46:57]3ffffd60:  00000026 3fff2024 00000000 3fff31b4  
[12:46:57]3ffffd70:  3ffffd78 00000000 3fffb700 40203d59  
[12:46:57]3ffffd80:  3fffb404 3fff2024 00000000 3fff989c  
[12:46:57]3ffffd90:  00000010 00000010 3fff928c 00000000  
[12:46:57]3ffffda0:  0000111c 00000000 3ffffdb0 00000000 <
[12:46:57]3ffffdb0:  00000000 3fff2ffc 3fff8d10 3fff31b4  
[12:46:57]3ffffdc0:  3ffffdc8 0000000d 616e6962 735f7972  
[12:46:57]3ffffdd0:  6f736e65 3fff0072 000003e9 40203f35  
[12:46:57]3ffffde0:  00000000 3fff7604 3fffb794 4021dd6c  
[12:46:57]3ffffdf0:  3fff8d10 000003e9 3fffb6fc 40201e36  
[12:46:57]3ffffe00:  6777ce2e 00000000 00015180 00000000  
[12:46:57]3ffffe10:  6777ce2e 00000000 00015180 00000000  
[12:46:57]3ffffe20:  00006777 f4240000 00000000 0000000c  
[12:46:57]3ffffe30:  0000f19f 3ffef4f0 c4a0f19f 0000f424  
[12:46:57]3ffffe40:  d5d0239c 00062acb 00000000 00000000  
[12:46:57]3ffffe50:  00000000 00062acb 000f4240 00000000  
[12:46:57]3ffffe60:  0000a8c0 33bbe717 00000004 3fff31b4  
[12:46:57]3ffffe70:  3fff8d10 3fff8d10 3fff31b4 40207ee4  
[12:46:57]3ffffe80:  00000239 00062acb c4a10239 0000f424  
[12:46:57]3ffffe90:  3ffe8368 00000000 00000239 40244b0a  
[12:46:57]3ffffea0:  0000f424 4023e67d 3fff47d4 3fff8fac  
[12:46:57]3ffffeb0:  00006777 f4240000 40237039 00062acb  
[12:46:57]3ffffec0:  d5d0239c 3fffff10 0001b278 00000000  
[12:46:57]3ffffed0:  00062acb d5d0239c 3fffff10 3fff4a5c  
[12:46:57]3ffffee0:  00004e7b 0000001c 3fff47d4 4023e695  
[12:46:57]3ffffef0:  00000000 00004430 e7ae147a 004ed88b  
[12:46:57]3fffff00:  181728f8 3fff1350 3fff4ac0 3fffa1ec  
[12:46:57]3fffff10:  3fff1978 00000001 3fffa220 4022ebf4  
[12:46:57]3fffff20:  00000000 fffff1f0 00000001 4022f0fc  
[12:46:57]3fffff30:  3fff4ac0 3fff1348 0000001c 00000000  
[12:46:57]3fffff40:  00000000 00004431 0978d4fd 004ed8b2  
[12:46:57]3fffff50:  060c2e36 00030003 07e90001 3fff31b4  
[12:46:57]3fffff60:  3fffb0c4 3fff31b4 3fffff80 3fff31b4  
[12:46:57]3fffff70:  3fffb0c4 00000002 3fff1390 4021c076  
[12:46:57]3fffff80:  00004431 3fff31b4 3fff8e8c 3fff1c18  
[12:46:57]3fffff90:  3fffb0c8 3fff46ac 3fff13c4 3fff1c18  
[12:46:57]3fffffa0:  3fffdad0 00000000 3fff1bec 3fff1c18  
[12:46:57]3fffffb0:  3fffdad0 00000000 3fff1bec 4023618c  
[12:46:57]3fffffc0:  feefeffe feefeffe 3fffdab0 40100719  
[12:46:57]<<<stack<<<
[12:46:57]`

Are there any known issues running this on ESP8266?

F1p, 2025-01-03:
What kind of 8266 are you using?

gekkekoe, 2025-01-03:
OOM = out of memory exception 
Your board is probably not having enough memory to run esphome latest version. I don't have 8266 boards to verify, and recommend to use an esp32(s3). Maybe @F1p can check if the current version is able to run on a 8266. If not then we probably need to remove support for it.

Bokenbergen, 2025-01-03:
Damn u guys are quick to reply!
I have tried with a few different ones. But they're 4MB boards! I will order a M5Stack sooner than later but this is what i have at home atm.

The exact model: https://www.az-delivery.de/en/products/d1-mini

And fyi @F1p MQTT version works on it.

F1p, 2025-01-03:
> OOM = out of memory exception Your board is probably not having enough memory to run esphome latest version. I don't have 8266 boards to verify, and recommend to use an esp32(s3). Maybe @F1p can [...]

I'm impressed you got logs - Yeah my 8266 boot loops, then stops responding - so think 8266 isn't possible anymore for this repo

Bokenbergen, 2025-01-03:
> I'm impressed you got logs - Yeah my 8266 boot loops, then stops responding - so think 8266 isn't possible anymore for this repo

Yup that's the exact same scenario i got over here. Keeps responding to ping but not much else :)
I wont bother u guys with this since i also believe the ESP32S3 is much more suitable for this application. But wanted to raise the issue so others won't have to deal with it! Thanks for providing this for the community!!!

gekkekoe, 2025-01-03:
cool, Will remove support for 8266 then. thanks for reporting

closed 2025-01-03 by gekkekoe