gekkekoe/esphome-ecodan-hp #110: Specific heat capacity for glycol

bwduncan, 2025-01-05:
Hi,

Just installed the latest update 216a4d4b73d69fd424b2494510c9a406d112094b and now I have working refrigerant condensing temp. Thank you! :partying_face: 

However I notice that the value used for the specific heat capacity is for pure water, not water/glycol mix, which I have. (I assumed that all pumps would run with glycol, but maybe installers in warm countries are allowed to assume that the pump will never be turned off when the outside temp is below freezing?). When the pump is commissioned, is there a way for installers to tell it that it's got glycol in it? Can we get that information out of the FTC again and use it to tweak the heat capacity value? If not, can we override it in the config?

Thanks!
Bruce

gekkekoe, 2025-01-05:
Split units don't use glycol, only monoblocks in cold countries I guess. Probably best to configure a constant, if defined use that else use fitted function. Here in the Netherlands is mostly pure water. It's used for the output estimate only btw.

Let me see what I can do

bwduncan, 2025-01-05:
Yes that's true, I don't know how much glycol the installers added to my system. I saw them pouring a 5L container into an upstairs rad but what was that mix and did they use more than one 5L container? Who knows! :laughing: 

Is Scotland a cold country? :laughing: 

Yes it's only the output estimate so not that important, but probably more accurate than the COP reported by the FTC!

gekkekoe, 2025-01-05:
See https://github.com/gekkekoe/esphome-ecodan-hp/commit/79510b34c99ca2bb1ee292cab51ed89577b4dd6b
Add the new option in main yaml to override:
```
ecodan:
  id: ecodan_instance
  uart_id: uart_ecodan
# uncomment line below and adjust the heat constant if your system uses a water + glycol mixture
#  specific_heat_constant_override: 3.9
```