gekkekoe/esphome-ecodan-hp #111: Cannot compile in ESP home with error: 'format_bin' was not declared in this scope

ChrisHomewood, 2025-01-05:
I Cannot compile in ESP home, I can provide the yaml file, but its basically a standard config but with the proxy feature enabled.
I think this error was introduced in Commit 41f9373 as it references a line that was added with that change, specifically:

> /config/.esphome/packages/224a9af1/confs/base.yaml: In lambda function:
> /config/.esphome/packages/224a9af1/confs/base.yaml:149:14: error: 'format_bin' was not declared in this scope
>        return format_bin(static_cast<uint8_t>(id(ecodan_instance).get_status().MRCFlag));
>                  ^~~~~~~~~~

Sadly I don't know enough about the code to work out how to rectify this error

gekkekoe, 2025-01-05:
update to latest esphome and do clean compile
pip install -U esphome

ChrisHomewood, 2025-01-05:
Thanks, I have now done that and it worked correctly... feel a bit silly about that one!!

I had temporarily commented out line 149 and it compiled and ran ok (with the status byte field blank on the web view), just recompiled and MRC status byte is now showing.

closed 2025-01-05 by ChrisHomewood