gekkekoe/esphome-ecodan-hp #113: Fault codes wrong?

andrazek, 2025-01-06:
I encountered an issue with a disconnected thermistor in my external unit. The error code recognized by the interface was incorrect compared to the one displayed on the FTC 5 unit. The FTC 5 unit showed a U4 code, while in HA, it was listed as U1 8105. Why?

gekkekoe, 2025-01-06:
when did you compile? did it include this commit 
https://github.com/gekkekoe/esphome-ecodan-hp/commit/ea497dce4a6d81564d08ad25025f8c138a9c6064 ?

gekkekoe, 2025-01-06:
65% is quite bad tbh :) what dbm? and how are your esp temps?

gekkekoe, 2025-01-06:
should be fixed now, I tested it locally with U4 and works as expected. Indeed the mask were wrong.

closed 2025-01-06 by gekkekoe

andrazek, 2025-01-06:
[image]
Maybe I can try relocating it a bit. Temperature doesn't look that bad. Thank you for fixing it so quickly!
[image]