gekkekoe/esphome-ecodan-hp #114: esp unavailable every 30 minutes

tzapu, 2025-01-06:
hi, got some kind of weird issue, it seems my module is going unavailable every 30 minutes, it does not seem to be restarting
[image]
[image]

i was wondering if you saw this issue before, i am running `confs/m5stack-atom-lite-proxy.yaml` with the latest code from github (but this was happening on earlier versions as well) or if you have some pointers on how i could start debugging this. i am sure it worked fine when i built it some time ago

thanks !

gekkekoe, 2025-01-06:
post esp temp graph and wifi db history graph

gekkekoe, 2025-01-06:
your temps seems a bit high, its the esp inside the unit ? what atom are you using ?

tzapu, 2025-01-06:
m5stack atom lite, been running like that for a few months, in a case on the wall. could add a few vent holes...
found the logs 
[image]

and signal
[image]

and temps working again, 45-47ºC
[image]

gekkekoe, 2025-01-06:
can you show more time around the reboot or show whole day wifi signal

tzapu, 2025-01-06:
thank you very much for all the hints, i will run through all of them and report back when i have enough data

it is not rebooting, seems to simply be disconnecting. i can't be sure, but i do not think i had the problem before as i ve only noticed it in the last couple of weeks.

gekkekoe, 2025-01-06:
The last link has a lot of tips for disconnects. I did not add any big changes last weeks, so its weird. you could try an older esphome, might have been introduced with the latest version

tzapu, 2025-01-06:
yeah, can t be from any of your changes for sure, maybe some esphome or home assitant changes, but i did not update your code for a long while until today, so the cause must be something else for sure

tzapu, 2025-01-07:
so went through everything, disabled all i could extra that i don t need, still keeps happening, maybe i should try a new esp module soon.

no idea if the loop time is ok or not?
also attached logs since a reboot up to the disconnect, apparently caused by ping not reposonding in 90s ...
[image]
[image]
[logs_heat-pump_run(1).txt](https://github.com/user-attachments/files/18330756/logs_heat-pump_run.1.txt)

gekkekoe, 2025-01-07:
Hmm everything looks fine. At this point I think it might be wifi connectivity. I see that the esp did not reboot so it must have been disconnected from another source. Can you check the following
* At your wifi accesspoint, does it record connection quality/disconnects ?
* do you have multiple accesspoints with the same name? It might pick the wrong one

Else try one of the atom s3 lite, that's what I use and they are quite good. You can also use a board with ethernet like heishamon to avoid wifi issues. At this point that's the main suspect.

F1p, 2025-01-07:
Are those loop times typical for esphome - if you have anything else to compare with?
My MQTT version of this project is <1ms loop time and there are some common libraries with esphome

WiFi strength is good, 70%ish - but i have heard of issues with Mesh networks and Atoms (on my project too) unless you pin to BSSID

gekkekoe, 2025-01-07:
looptimes varies, depends on if it needs to render stuff. He also has logs attached so looptimes will be a bit higher.
[image]

gekkekoe, 2025-01-07:
I have one idea left, you mentioned using proxy. Can you test without the melcloud attached?

gekkekoe, 2025-01-07:
If that works, then we know its that combination.
If thats the case, please enable uart logging 
by adding 
```
    # debug:
    #   direction: BOTH
    #   dummy_receiver: false
    #   after:
    #     delimiter: "\n"
    #   sequence:
    #     - lambda: UARTDebug::log_hex(direction, bytes, ' ');
```
I will add the commented logging in your yaml.
uncomment and build locally, could be that we do not handle some response correctly (but I would expect a crash/reboot in that case)

tzapu, 2025-01-08:
hey guys, i tried most of what you recommended to no avail (thank you again for all the suggestions, super helpful) and in the end i got the soldering iron out and installed a new lolin s3 module i had. since then i ve had a 10h run without any disconnection, which is an absolute massive improvement. i ll leave it running for a bit in this barebones config, then start adding back proxy, removing bssid, etc. will keep reporting on this. thanks

gekkekoe, 2025-01-08:
Ah so proxy failed also with atom? Okay then it's probably that hardware and your network combination :( So we can rule out the software.

I use the atom S3 lite and it works great here in my network.

Why do you still need melcloud btw?

DamijanD, 2025-01-08:
My atom works with proxy/melcloud without any issues from the start. So in general proxy works OK.