gekkekoe/esphome-ecodan-hp #120: action fails to set climate temp

bwduncan, 2025-01-19:
Hi,

Happy new year!

Has something changed in the climate component? I've got an HA automation which turns the heat pump on and sets the zone 1 climate temperature when the electricity rate goes off peak, but it's not working:

```
[04:00:22][D][climate:011]: 'Zone 1 Climate' - Setting
[04:00:22][D][climate:040]:   Target Temperature: 38.55
[04:00:22][D][climate:396]: 'Zone 1 Climate' - Sending state:
[04:00:22][D][climate:399]:   Mode: HEAT
[04:00:22][D][climate:401]:   Action: IDLE
[04:00:22][D][climate:419]:   Current Temperature: 20.00°C
[04:00:22][D][climate:425]:   Target Temperature: 38.55°C
[04:00:27][I][ecodan.component:040]: Could not publish state of sensor 'status_three_way_valve_2' with value: '1'
[04:00:27][I][ecodan.component:040]: Could not publish state of sensor 'status_water_pump_3' with value: '0'
[04:00:27][I][ecodan.component:018]: Could not publish state of sensor 'status_mixing_valve' with value: '4.000000'
[04:00:27][I][ecodan.component:018]: Could not publish state of sensor 'mixing_valve_step' with value: '0.000000'
[04:00:30][I][ecodan.component:018]: Could not publish state of sensor 'status_operation' with value: '2.000000'
[04:00:30][I][ecodan.component:018]: Could not publish state of sensor 'status_heating_cooling_z2' with value: '1.000000'
[04:00:30][D][climate:396]: 'Zone 1 Climate' - Sending state:
[04:00:30][D][climate:399]:   Mode: HEAT
[04:00:30][D][climate:401]:   Action: HEATING
[04:00:30][D][climate:419]:   Current Temperature: 20.00°C
[04:00:30][D][climate:425]:   Target Temperature: 38.55°C
```

For reference the (redacted) automation looks like:

```
triggers:
  - entity_id:
      - binary_sensor.octopus_energy_target_night_heating
    to: "on"
    trigger: state
    enabled: true
  - trigger: state
    entity_id:
      - sensor.octopus_energy_electricity_redacted_current_rate
    to: null
    enabled: false
  - trigger: numeric_state
    entity_id:
      - sensor.octopus_energy_electricity_reedacted_current_rate
    below: 0.16
conditions:
  - condition: state
    entity_id: binary_sensor.octopus_energy_target_night_heating
    state: "on"
    enabled: false
  - condition: numeric_state
    entity_id: sensor.octopus_energy_electricity_redacted_current_rate
    below: 0.13
    enabled: true
actions:
  - type: turn_on
    device_id: f24459727e584829e84a18020c407e91
    entity_id: e0dabae46282508887590120936bcf84
    domain: switch
  - action: climate.set_temperature
    metadata: {}
    data:
      temperature: >
        {{ min([60,max([20,20 +
        2.5*(20-100*float(states('sensor.octopus_energy_electricity_redacted_current_rate')))])])
        }}
    target:
      entity_id: climate.ecodan_heatpump_zone_1_climate
  - action: switch.turn_on
    metadata: {}
    data: {}
    target:
      entity_id:
        - switch.laptop
        - switch.dehumidifier_2
mode: single
```

Any ideas? It only seems to be a problem with this automation. I can set the temperature manually in the UI and it works, but not at 4am 🤣 

THanks

gekkekoe, 2025-01-19:
The climate will only read back state from Heatpump at 60s interval max. That has not changed. Which date did you build and flash? So I assume this was working before?

I adjusting polling recently, so could be a timing. But my automations seems to be working fine :/

You can test this action/automation within HA. It will execute without checking the conditions

Then look at the traces to see why it fails. I think it's timing, since we poll a bit faster now

bwduncan, 2025-01-19:
Hi, 

I'm out now, typing this for the third time on mobile. Bloody phones! 

I pulled built and flashed last night before it did it again. I keep pretty close to main branch.

Didn't know about delay. Would it be ok to send the same temp twice in 60s? I don't think I'm doing that though. 

Traces don't look weird... I'll upload a screenshot when I get home and my phone stops eating my text...

Thanks!

gekkekoe, 2025-01-19:
I do see a climate log within 2 seconds.
you can send as many as you want. But it will only accept a new one after 60s after the last change.
But I did just test my automation, it also sets the climate and it seems to work. Don't see why it would not work for you

gekkekoe, 2025-01-19:
So from your logging I do see that it went from IDLE to HEATING, whats the problem then btw? A new setpoint is not set ?

bwduncan, 2025-01-19:
Yes exactly. I have an entity history card (graph) which shows a spike to the correct temp but then immediately back to the previous setting. The heat pump turns on but nothing happens because the flow temp is still set to 20.

gekkekoe, 2025-01-19:
Ah yeah timing. but this always worked like this. 

try 60a delay between each climate set. should work

gekkekoe, 2025-01-19:
yup, but you know that you can test the actions without the conditions? (in automations, and select the automation, there's a sub menu where you can run the actions)

bwduncan, 2025-01-19:
Thanks, yes I tried that but it worked every time. I think the problem is two triggers so just running the actions doesn't reproduce the problem. 

The 1300 run worked btw so likely that was the problem

gekkekoe, 2025-01-19:
ah cool, yeah you probably got lucky. It always worked like this. Maybe the adjusted timing triggered it for you

bwduncan, 2025-01-19:
Well, that's two successful triggers in a row. I think it's safe to close this now. I was tempted to make a PR to document the fact that only one update can be made within the refresh interval, but I couldn't find a convenient place to put it. On the other hand, is there any locking mechanism which will allow a second update to override the first?

gekkekoe, 2025-01-19:
I disabled it because it was causing issues.
When we send a cmd (new setpoint) it takes some time to get the confirmed value from the Heatpump.

If we did not get the confirmed value, but still send a new value, it will lead to issues.

The UI has the latest state, but then the first cmd is processed and returned. So now the UI is in an inconsistent state (displaying new setpoint, but Heatpump has first setpoint).

Otoh, it does not really make sense to spam a climate with so many updates. It's partly due to esphome and handling via the UI. I haven't found a smart/clever way to get around it.

gekkekoe, 2025-01-19:
I will convert this to discussion for future reference