gekkekoe/esphome-ecodan-hp #122: Could not publish state of sensor

MrBretze, 2025-01-24:
Hello, I have installed this implementation with an Atom S3 Lite and I have a lot of info with "Could not publish state of sensor 'xxxxxx' with value: 'xxxxx'"
I give you my logs, I have enabled the debug if it can help you, and a screen of my HA.
I have an Ecodan ERSD-VM6D

[image]

[logs_ecodan-pump_logs.txt](https://github.com/user-attachments/files/18541715/logs_ecodan-pump_logs.txt)

gekkekoe, 2025-01-24:
You are missing a lot of sensors, did you use the right template? default one

MrBretze, 2025-01-24:
Ogh sorry... I have pasted the code from the REDME.md (main page) not the yaml in the repository... I try why this one and tell you if it works (probably 😅) ! 😊

MrBretze, 2025-01-24:
And it works !!!! Thank you and sorry 😁

closed 2025-01-24 by MrBretze

gekkekoe, 2025-01-24:
np, will fix the readme, but did not think ppl would copy from there