gekkekoe/esphome-ecodan-hp #146: confs/status_led_rgb.yaml breaks compling

Bokenbergen, 2025-03-26:
Hello,

Not sure if this is due to you or nielsnl68's code beeing outdated. But when i enable the rgb yaml i get these errors:
```
INFO ESPHome 2025.3.2
INFO Reading configuration /config/esphome/xxx.yaml...
INFO Updating https://github.com/gekkekoe/esphome-ecodan-hp/@main
INFO Updating https://github.com/gekkekoe/esphome-ecodan-hp.git@main
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1057, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1035, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 1095, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 949, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 939, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 835, in validate_config
    target_platform = core_config.preload_core_config(config, result)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/core/config.py", line 231, in preload_core_config
    if _is_target_platform(domain):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/core/config.py", line 192, in _is_target_platform
    return get_component(name, True).is_target_platform
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/loader.py", line 205, in get_component
    return _lookup_module(domain, exception)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/loader.py", line 181, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/c0810ba7/esphome/components/wifi/__init__.py", line 96, in <module>
    cv.Required(CONF_STATIC_IP): cv.ipv4,
                                 ^^^^^^^
AttributeError: module 'esphome.config_validation' has no attribute 'ipv4'
```

Bokenbergen, 2025-03-26:
Guess confs/esp32s3-led.yaml should be used instead?

gekkekoe, 2025-03-26:
yes for atoms use that one

Bokenbergen, 2025-03-27:
I see, works great now! 
Thanks for cleaning up the repo, wouldn't have gotten confused with the current files 😅