gekkekoe/esphome-ecodan-hp #149: Cannot compile

Rvh91, 2025-04-01:
**Describe the bug**
I had previously copied  the base, labels-en, and zone1 yaml files, such that I could change the labels manually to something that matched with the rest of the entity labeling in Home assistant. However, recently I have not been able to update the esphome version anymore (it is now on 2024.12.4, compiled on january 19th this year). 
I figured there might have been a change in the component (which was still always pulled from your repository), that might no longer match with the configuration yamls. 
However, today I changed back over the configuration files to also be pulled from the repository, and it still does not compile. I'm a bit lost on what the issue is exactly.

**To Reproduce**
Steps to reproduce the behavior:
1. I used the following yaml:

``` 
esphome:
  name: ${node_name}
  comment: "Control the Zehnder ComfoAir E400 Fanspeed via pwm, and Ecodan via uart"

  on_boot:
      then:
        - fan.turn_on: 
            id: mechanical_ventilation

esp32:
  board: esp32dev
  framework:
    type: arduino

packages:
    remote_package:
    url: https://github.com/gekkekoe/esphome-ecodan-hp/
    ref: main
    #refresh: always
    files: [ 
            confs/base.yaml,        # required
            confs/request-codes.yaml,   # disable if your unit does not support request codes (service menu)
            confs/zone1.yaml,
            #confs/zone2.yaml,      # enable if you want to use zone 2
            ## enable label language file
            confs/ecodan-labels-en.yaml,
            #confs/ecodan-labels-nl.yaml,
            confs/server-control.yaml,
            #confs/debug.yaml,
            ## enable this to monitor WiFi status with ESP in-built LED
            #confs/status_led.yaml,
           ]
  
### new
external_components:
  - source: github://gekkekoe/esphome-ecodan-hp@main
    components: [ ecodan ]
    refresh: always

uart:
  - id: uart_ecodan
    tx_pin: GPIO17
    rx_pin: GPIO5
    baud_rate: 2400
    parity: EVEN
    rx_buffer_size: 1024

#define the ecodan
ecodan:
  id: ecodan_instance
  uart_id: uart_ecodan

```
compiling then fails with the errors shown below:

**Expected behavior**
I had expected to install an update without issue.

**Heatpump model**
ecodan erst20d vm2d (although I doubt that matters for this issue)
[... 38 lines cut ...]
src/esphome/components/ecodan/commands.cpp:246:121: error: no matching function for call to 'esphome::ecodan::ServiceCodeRuntime::ServiceCodeRuntime(<brace-enclosed initializer list>)'
         ServiceCodeRuntime{Status::REQUEST_CODE::TH6_2_PHASE_PIPE_TEMP, true, 0, std::chrono::steady_clock::time_point{}},
                                                                                                                         ^
src/esphome/components/ecodan/commands.cpp:234:12: note: candidate: 'esphome::ecodan::ServiceCodeRuntime::ServiceCodeRuntime()'
     struct ServiceCodeRuntime {
            ^~~~~~~~~~~~~~~~~~
src/esphome/components/ecodan/commands.cpp:234:12: note:   candidate expects 0 arguments, 4 provided
src/esphome/components/ecodan/commands.cpp:234:12: note: candidate: 'constexpr esphome::ecodan::ServiceCodeRuntime::ServiceCodeRuntime(const esphome::ecodan::ServiceCodeRuntime&)'
src/esphome/components/ecodan/commands.cpp:234:12: note:   candidate expects 1 argument, 4 provided
src/esphome/components/ecodan/commands.cpp:234:12: note: candidate: 'constexpr esphome::ecodan::ServiceCodeRuntime::ServiceCodeRuntime(esphome::ecodan::ServiceCodeRuntime&&)'
src/esphome/components/ecodan/commands.cpp:234:12: note:   candidate expects 1 argument, 4 provided
src/esphome/components/ecodan/commands.cpp:251:110: error: no matching function for call to 'esphome::ecodan::ServiceCodeRuntime::ServiceCodeRuntime(<brace-enclosed initializer list>)'
         ServiceCodeRuntime{Status::REQUEST_CODE::FAN_SPEED, false, 0, std::chrono::steady_clock::time_point{}}
                                                                                                              ^
src/esphome/components/ecodan/commands.cpp:234:12: note: candidate: 'esphome::ecodan::ServiceCodeRuntime::ServiceCodeRuntime()'
     struct ServiceCodeRuntime {
            ^~~~~~~~~~~~~~~~~~
src/esphome/components/ecodan/commands.cpp:234:12: note:   candidate expects 0 arguments, 4 provided
src/esphome/components/ecodan/commands.cpp:234:12: note: candidate: 'constexpr esphome::ecodan::ServiceCodeRuntime::ServiceCodeRuntime(const esphome::ecodan::ServiceCodeRuntime&)'
src/esphome/components/ecodan/commands.cpp:234:12: note:   candidate expects 1 argument, 4 provided
src/esphome/components/ecodan/commands.cpp:234:12: note: candidate: 'constexpr esphome::ecodan::ServiceCodeRuntime::ServiceCodeRuntime(esphome::ecodan::ServiceCodeRuntime&&)'
src/esphome/components/ecodan/commands.cpp:234:12: note:   candidate expects 1 argument, 4 provided
*** [.pioenvs/fancontrol-eth/src/esphome/components/ecodan/commands.cpp.o] Error 1
========================= [FAILED] Took 26.87 seconds =========================
```

gekkekoe, 2025-04-01:
```
esp32:
  board: esp32dev
  framework:
    type: arduino
```

I've switched to esp-idf long time ago.
copy relevant parts from:
https://github.com/gekkekoe/esphome-ecodan-hp/blob/main/confs/esp32.yaml

also why disable refresh in
```
packages:
    remote_package:
    url: https://github.com/gekkekoe/esphome-ecodan-hp/
    ref: main
    #refresh: always
```

Rvh91, 2025-04-01:
Thank you that seems to have done the trick (at least it's still compiling!). Apologies for the inconvenience. The switch to esp-idf must have been somewhere after the 14th of januari I'm assuming? 

the commented refresh always is probably a remnant of me trying various things in order to get it working again.

gekkekoe, 2025-04-01:
It was switched in august (2024)
https://github.com/gekkekoe/esphome-ecodan-hp/commit/efc44bcb11235b45c0940325bfcee12ffdc2f1ab