gekkekoe/esphome-ecodan-hp #151: Refrigerant

Duntch144, 2025-04-16:
**Describe the bug**
Refrigerant reported is R410A but HP uses R32

**Expected behavior**
Should report R32

**Screenshots**

[image]

**Heatpump model**
PUZ- SHWM 140

gekkekoe, 2025-04-16:
are you able to enable log and attach it here? I'm looking for the response of the 0xC9 message.

Duntch144, 2025-04-16:
CMD { .Hdr { FC, 5B, 02, 7A, 10 } .Payload { C9, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00 } .Chk { 50 } }

Firmware 18.01 
FTC6

gekkekoe, 2025-04-16:
the reply of this msg please :)

Duntch144, 2025-04-16:
CMD { .Hdr { FC, 5B, 02, 7A, 10 } .Payload { C9, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00 } .Chk { 50 } }
00:12:58	[D]	[text_sensor:064]	
'Operation Mode': Sending state 'Off'
00:12:58	[D]	[sensor:093]	
'controller_version': Sending state 3.00000  with 0 decimals of accuracy

Duntch144, 2025-04-17:
Here is the line : FC 7B 02 7A 10 C9 03 00 01 00 14 03 00 01 01 00 00 00 00 00 00 13

gekkekoe, 2025-04-17:
Thanks, mines reports the same, hmm weird that your unit reports 0 as byte 10. But I probably need to collect more samples

gekkekoe, 2025-04-21:
It will be disabled next release since it does not work as expected

closed 2025-04-21 by gekkekoe