gekkekoe/esphome-ecodan-hp #158: DHW setpoint changes

Bokenbergen, 2025-05-23:
I believe since a few commits back there is some bug setting my dhw setpoint to a lower value. Have had this happen three times now and I can replicate it by recompilig and flashing. It happens during normal operation and on esphome install.

Can I somehow log what changes this setpoint? 

Guessing it started with commit:
671367a0eb9967f30c68e66ffdc835787ccd0347

gekkekoe, 2025-05-23:
I have dhw at 20c for 6 hours now, and it does not change back.

You can enable logging (new switch) and log logs, and maybe post -1m and +1m log part at the moment it changes back

You are the 2nd person reporting this so, its so weird that mines does not behave like this. we need to find the trigger that changes it, could be some unrelated command, and FTC tries to 'correct' the faulty value (I am guessing here) Try to see at that time if operating mode changes

WesSec, 2025-05-23:
Following and contributing, i have behavior like this, just the opposite, when i set the value to 20, at a random interval it get set to 40 again

logs i posted to tweakers, time of event 12:45:48: https://gist.github.com/WesSec/22ba7c29c03cc8bf459e278bc1182f36

Bokenbergen, 2025-05-23:
Figured it out. Only automation triggers this issue.
I used to send action to device, now I've changed it to entity, which mitigates the issue.
Not sure if this is how its suppose to work or not, but my old automation has worked just fine earlier!

How to trigger it:
[image]

How to fix it:
[image]

gekkekoe, 2025-05-23:
> Following and contributing, i have behavior like this, just the opposite, when i set the value to 20, at a random interval it get set to 40 again
> 
> logs i posted to tweakers, time of event 12:45:48: https://gist.github.com/WesSec/22ba7c29c03cc8bf459e278bc1182f36
> 
> FWIW: S3 Lite, no proxy setup

Thanks for the logs.
[12:45:32][D][sensor:093]: 'DHW Setpoint Value': Sending state 20.00000 °C with 1 decimals of accuracy
...
[12:45:48][D][sensor:093]: 'DHW Setpoint Value': Sending state 40.00000 °C with 1 decimals of accuracy

I don't see any set action from the atom between the 20c and 40c. So It must happen outside my component.
Do you have wireless remote thermostats?

gekkekoe, 2025-05-23:
> These sensors does not write to log book. which is one of the reasons it took me a while to figure this out, perhaps that would be a nice thing to add?
> 
> I'm pretty sure however that triggering it on the device sets all three to whatever value you have in the automation.

in the climate history, you can see the set of the target temp, see my screenshot above, where I set mines to 20c manually to test if it gets reset. in your case probably all your climate will have a set to 19c at that point in time

gekkekoe, 2025-05-24:
Resideo HCC100 should not cause the issue. Are you using remote thermostats from mitsubishi ? (PAR-XX)

WesSec, 2025-05-24:
Got some logs for you. I do not have any external thermostats from Mitsubishi, only the ecodan unit in the closet and a thing on the roof.

As mentioned on tweakers. it might be a result of checks done by the MRC. I put the value to 25, activated the MRC (no changes) and directly the value was set back to 40.

Timestamps:

```
11:49 base value: 55 from previous setting
11:50:02 set to 25
11:53:something i lit up the panel on the ecodan, did not change anything
11:53:39: back to 40 again

```
Logs:

[logs_ecodan_logs (1).txt](https://github.com/user-attachments/files/20422264/logs_ecodan_logs.1.txt)

gekkekoe, 2025-05-24:
Very helpful. I managed to reproduce the issue.
It seems that any change on the display (MRC) writes back the entire state to the ecodan. Since MRC can only handle min 40c, it will write 40c to the ecodan. 

I'm afraid I cannot do anything against this. Even considering reverting the allowing lower than 40c setting to the dhw climates.

gekkekoe, 2025-05-25:
https://github.com/gekkekoe/esphome-ecodan-hp/commit/1ba7fb06b4596758901290c5e4bbff7a2409f90c

closed 2025-05-25 by gekkekoe