gekkekoe/esphome-ecodan-hp #162: Missing part in confs/esp32.yaml

x1marc, 2025-06-11:
I wanted to implement the entire project on an esp32c3, and when compiling it into esphome and creating the .bin file, this or something similar is missing:

esphome:
name: ${name}
friendly_name: ${friendlyName}
platformio_options:
board_build.flash_mode: dio

I was able to fix this error by copying it to my local system.

I also created a German translation:
ecodan-labels-de.yaml

How do I submit this to you?

Regards, Marc

gekkekoe, 2025-06-11:
Create pull request? or if you don't know how, just attach yaml here

gekkekoe, 2025-06-11:
please drag the yaml in the msg, else spacing gets lost

gekkekoe, 2025-06-11:
great thanks a lot ! I will add it and add de to build pipe line. Then we can have .de bins as well.

gekkekoe, 2025-06-11:
https://github.com/gekkekoe/esphome-ecodan-hp/releases
I've fixed the esp32.yaml and added the de yaml in the build. Thanks for your contribution.

closed 2025-06-11 by gekkekoe